carla-simulator/carla #1349: Waypoints not positioned at the centre of lanes

pravinblaze, 2019-03-07:
There seems to be a problem in how the waypoints are positioned on the lane in some road segments.
This is causes problems while trying to localise actor positions relative to the lane centre since the waypoints are not always at the visible centre of lanes.

marcgpuig, 2019-03-12:
@pravinblaze 
Is this in Town03?

nishanthjois, 2019-03-18:
Yes, it is in Town03.

nishanthjois, 2019-03-29:
Any update on this?

akki6843, 2019-04-01:
Closing this as the issue is fixed with latest build, spawning of props is now correct (i.e. on Sidewalk) in Town03 and Town05.
Here is the screenshot of vending machine spawning on the sidewalk.
[image]

closed 2019-04-01 by akki6843