carla-simulator/carla #1359: Client.get_world() failed in Carla 0.9.3 and Carla 0.9.4

zhang-xinxin, 2019-03-10:
My OS is Ubuntu 16.04 and I ever tested many Carla version from 0.8.1 to 0.9.2, all these version worked well.

Last week I updated Carla from 0.9.2 to 0.9.3, compiled release. I launch the server side successfully, and I can see the town03 in window. 
`./CarlaUE4.sh /Game/Carla/Maps/Town03`

But when I run the client script, e.g. manual_control.py,  it failed with an RuntimeError
Traceback (most recent call last):
File "manual_control.py", line 798, in <module>
    main()
File "manual_control.py", line 842, in main
    game_loop(args)
File "manual_control.py", line 768, in game_loop
    world = World(client.get_world(), hud, args.filter)
RuntimeError: std::bad_cast

I go back to version 0.9.2, and launch the server and client again, they still works well. I can even run the server side in version 0.9.2 and run the client side in version 0.9.3.  But I cannot run the server and client in version 0.9.3. 

Today I downloaded the version 0.9.4 and try, but the RuntimeError happened again.
I try to build the simulator from source, but the Error is the same.

Anybody has the same problem? What should I do to fix it?

Thanks

nsubiron, 2019-03-11:
That usually happens when the version of the server and the client are different, in this case, on 0.9.3 we changed slightly the RPC call of the `get_world` request.

You are probably using an old version of the client API (Python API). Make sure you use the egg files that come with the release package, if you have installed the carla Python module in your machine, you should reinstall.

zhang-xinxin, 2019-03-12:
@nsubiron Thank you very much! I solved the problem finally and it was due to the incompatibility of egg version. 

I re-installed the new egg file, but the client connection failed again. So I printed and double checked the system path, and found the Carla 0.9.2 directory still included in the path, because it was set in $PYTHONPATH in .bashrc file.
After the system path cleaned, the simulator can work well.

closed 2019-03-12 by zhang-xinxin