carla-simulator/carla #1360: Question about depth image converted point cloud?

JackRen88, 2019-03-10:
Hello,everyone!
I wrote a script to convert depth and RGB images into point clouds.I have a confused question that when the car is traveling straight, a series of depth images are converted into corresponding point clouds, which can be spliced together very well, but when the car makes a turn, the converted point clouds are not well stitched together, such as Figure:
[image]
I don't know where the problem is, please help me solve this problem.
point cloud file format:out.pcd

[output.pcd.zip](https://github.com/carla-simulator/carla/files/2949188/output.pcd.zip)

closed 2019-05-16 by stale[bot]

marcgpuig, 2019-05-20:
Hi @JackRen88 
Looks like there is something wrong on your rotation code, can you provide more information on how are you doing it?

sebastianvarghese, 2023-01-05:
Hi @JackRen88 I was trying to convert carla depth and RGB images to point clouds. Could you please share your scripts.