carla-simulator/carla #1363: red arrow in rendered camera image [Carla 0.9.4]

dianchen96, 2019-03-10:
Hi,

When I save the images rendered from an rgb camera, I found that there is always a red arrow in the image, seemingly pointing to the car's current velocity when the car is moving. Can anyone point me out how to remove this arrow in the image? Much appreciated!

[image]
[image]

slowergmajor, 2019-03-11:
Hi，
I don't know if it's right. According to my experience, this arrow appears in the official "automatic_control.py". Trying to use the "manual_control.py" arrow will not appear.

dianchen96, 2019-03-11:
@slowergmajor thanks for your reply! I am using my own script to collect the images. If the official automatic_control.py also has the arrow then it seems to have something to do with agent. would love to have the authors confirm on this.

aravindSwamy94, 2019-03-11:
@dianchen96 : u have to disable the debug in local_planner.py if you are running the automatic_control.py

in local_planner.py --->  run_step(self, debug=False)

This will remove the debug lines. I hope this helps

dianchen96, 2019-03-11:
thank you! this fixed the issue

closed 2019-03-11 by dianchen96

qhaas, 2019-06-20:
Why is this debug flag enabled by default?