carla-simulator/carla #1366: Image Classification on the RGB images collected from camera

BoChang11, 2019-03-11:
Hi,

I am trying to classify the images collected from the ```RGB camera```. 
For example, <br /> **Class A** : Frame with moving vehicle in front of ego vehicle and frame <br /> **Class B** : Frame with no vehicle in front of ego vehicle.

Is there any pre-processing steps need to be followed before dumping the RGB images to local disk ? Or can I use the RGB images directly for this kind of classification problem?

BR.

nsubiron, 2019-03-12:
Hi @BoChang11,

You can save them just as they are generated, with
```py
image.save_to_disk('filename.png')
```
In fact, we don't provide any post-processing for RGB images.

BoChang11, 2019-03-13:
Thanks for the reply. 
One more question : Do I need to convert the pixel co-ordinates from 3D to 2D before training the classification model?

BR

closed 2019-05-19 by stale[bot]