carla-simulator/carla #1369: How to set mini-displays on the main screen?

robinalen, 2019-03-12:
OS: Linux Ubuntu
Version: 0.9.3

I have set multiple camera sensors to the vehicle but I can only use TAB to change between sensors. I want to make those cameras like this:
[image]

closed 2019-03-12 by robinalen

vishal-prgmr, 2019-06-07:
Did you figure this out?

robinalen, 2019-06-07:
Hi dude, you can refer to the manual_control.py. In order to get those mini-screen, you have to modify the cameraManager class to add camera sensors you want and turn on listening to transmit sensors’ data. Also don’ forget to adjust pygame objects to stick them to the main screen.

vishal-prgmr, 2019-06-07:
Implementation would really save a lot of time. Do you have one? Dont wanna mess with the infiltratable manual_control.py .

robinalen, 2019-06-07:
Unfortunately, I have finished the Carla program last month and those files are inaccessible for me anymore:<

AmirHarvey, 2020-03-29:
Hello, i am trying to do as the attached photo in the beginning. Can you help?

darld, 2021-07-06:
@carla-admin. Can you provides details about the code change of the manual_control.py?