carla-simulator/carla #1370: Obtain cropped mesh of the scene

luigui2906, 2019-03-12:
Hi,

I am writing to know if there is any easy way to obtain a crop or a segment of the mesh of the scene surrounding the vehicle. I would like to have this because I am testing an algorithm for scene reconstruction from pointcloud using the simulated lidar and I would like to compare my results against the actual real mesh of the scene. Having the possibility of extract the mesh of a portion of the scene, (equivalent to what it has been scanned by the simulated lidar) would be awesome.

Thanks in advance.

marcgpuig, 2019-03-12:
Hi @luigui2906, thanks for the interest on Carla.
This is something we want to make in the future but currently, we don't have anything like that.
You can modify the server and create some kind of sensor that does such a thing.

luigui2906, 2019-03-14:
Hi @marcgpuig. Ok. Thanks for the answer.

closed 2019-05-20 by stale[bot]