carla-simulator/carla #1372: No modul named carla - Windows experimental built 0.9.3

dieteekanne2, 2019-03-12:
Hi, 
when I try to launch any of the example scripts (eg 'manual_control.py') I get the following error:

C:\Users\...\CARLA_0.9.3>python manual_control.py
Traceback (most recent call last):
  File "manual_control.py", line 65, in <module>
    import carla
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\__init__.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\libcarla.py", line 7, in <module>
  File "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg\carla\libcarla.py", line 6, in __bootstrap__
  File "C:\Users\maxhe\Anaconda3\lib\imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: The named modul can't be found

Then I tried to switch to a venv with pygame and numpy installed, the error changed to:

C:\Users\...\CARLA_0.9.3>conda activate CarlaVenv

(CarlaVenv)C:\Users\...\CARLA_0.9.3>python manual_control.py
Traceback (most recent call last):
  File "manual_control.py", line 65, in <module>
    import carla
ImportError: No module named 'carla'

But here I'm stuck. I tried the experimental 0.9.4 too, same problem.

Thanks!

marcgpuig, 2019-03-12:
Hi @dieteekanne2 
Looks like @SAmmarAbbas get it working creating a new Anaconda environment, see #1325.

SAmmarAbbas, 2019-03-12:
@dieteekanne2 the second error is because the file "PythonAPI\carla-0.9.3-py3.7-win-amd64.egg" is not added to your path while running the start of the script. This could be because you are not using Python 3.7 most probably or could be something else. As a result, it cannot find 'carla'

dieteekanne2, 2019-03-13:
thanks @marcgpuig and @SAmmarAbbas for your help. Now it works, here's a short description if anyone else has the same problem.
Apparantly Anaconda and the venv caused the problem for me. I tried it with new Anaconda environments, but the problem persisted. Then I realized that Anaconda was downgrading my python version to 2.7 as soon as I installed pygame. I tried to create a new virtual environment using just python, but there I run into problems with the SSL module. After following the instructions from [this comment](https://github.com/pypa/virtualenv/issues/1139#issuecomment-453865723) I could use pip, install numpy and pygame and it finally works. Thanks!

twak, 2019-03-13:
Can confirm, installed Anaconda 64bit with Python 3.7, opened the Anaconda Prompt, navigated to the Carla directory with the python scripts and  `python spawn_npc.py -n 80` worked.

closed 2019-05-19 by stale[bot]

autonomobil, 2019-08-19:
I had a similar issue in a conda enviroment. This was my solution:
 
- locate .egg-files in `Carla/PythonAPI/carla/dist/`
- change/install python version in conda enviroment which fits closest to an .egg-file (for me carla-0.9.6-py3.5-linux-x86_64.egg -> 3.5)
- delete all other .egg-files (just for tidiness)
- extract your .egg-file, e.g. carla-0.9.6-py3.5-linux-x86_64
- create `setup.py` inside the folder (carla-0.9.6-py3.5-linux-x86_64) with the following content:

```
from distutils.core import setup
setup(name='carla',
      version='0.9.6', #doesn't matter I guess
      py_modules=['carla'],
      )
```
- then you can install the carla-python-package via pip: 
`pip install -e ~/Carla/PythonAPI/carla/dist/carla-0.9.6-py3.5-linux-x86_64`

That solved it for me. I don't even need the egg-path-appending-stuff, which is happening at the beginning of every python-script, just `import carla.`

nuomizai, 2020-05-08:
> I had a similar issue in a conda enviroment. This was my solution:
>
> * locate .egg-files in `Carla/PythonAPI/carla/dist/`
> * change/install python version in conda enviroment which fits closest to an [...]

thank you, this works for me! However, after I successfully import carla and test a reinforcement learning agent trained by carla, an error occurs as:
ModuleNotFoundError: No module named 'carla.driving_benchmark'

ruelj2, 2021-06-23:
> I had a similar issue in a conda enviroment. This was my solution:
>
>     * locate .egg-files in `Carla/PythonAPI/carla/dist/`
>
>     * change/install python version in conda enviroment which fits [...]

I would rather:
- Create a setup.py in Carla/PythonAPI directly with `sudo bash -c 'echo "from distutils.core import setup;setup(name="carla", version="0.9.10", packages=["carla"])" > Carla/PythonAPI /setup.py'`
- Then `pip install Carla/PythonAPI/`