carla-simulator/carla #1373: Is there a way to add car headlights ?

FrancoisMasson1990, 2019-03-12:
Hello,

If I want to have car headlights, what is the best way to add in the car assets ?

TheNihilisticRobot, 2019-03-12:
Hi @FrancoisMasson1990 

The quickest way to achieve this is opening the blueprint of the vehicle you need to have lights on and adding two point light components It is also a bit of a workaround, theese lights won't be able to turn on or off, and since the point light only lights in a cone the lights won't look on from outside of the vehicle.

FrancoisMasson1990, 2019-03-13:
Is it possible to implement it trough the PythonAPI or this need to be done in the Unreal Editor ?

TheNihilisticRobot, 2019-03-13:
It is not possible at the moment to add lights from the pythonAPI but since that would be pretty useful I'll note it as a feature request and We'll take it in consideration as such.

FrancoisMasson1990, 2019-03-13:
Appreciate. What would be nice is to be able ,for example, to insert spotlight and attach it to the player and choosing the orientation + light intensity

p-schulz, 2019-04-20:
As inspiration: I did exactly the same with my vehicles, just added two spotlights at the front of the car (with some light blue tint). There are also small spheres inside the head and tail lights with an emissive material, since the spotlight isn't a glowing volume by itself. For performance reasons they don't cast shadows in the scene. Depending on the time of day they are turned on and off, this is what they look like in the editor in Town04:

[image]

and when the simulation is running:

[image]
[image]

FrancoisMasson1990, 2019-04-20:
thanks @p-schulz ! 
Can you provide the file .uasset generated for the vehicle ?

EmilioOldenziel, 2019-05-03:
Hi @p-schulz, 
That looks really  good! Do you by any chance have your settings somewhere that I could try?
Thanks,
Emilio

closed 2019-07-09 by stale[bot]

revikiran1991, 2020-02-23:
@p-schulz I ma struggling a bit to switch ON streetlights and vehicle lights. Can you please help me by sharing your  project settings.