carla-simulator/carla #1375: Lidar setting manual_control.py

Atlunch, 2019-03-12:
Hello everyone,

I would like to have a 360 Lidar point clound visible on the screen.

I added the following code to the manual_control.py but it dosen't affect the lidar.

What I'm doing wrong?

client = carla.Client(args.host, args.port)
client.set_timeout(2.0)
world = client.get_world()
bluelib = world.get_blueprint_library()
lidarbp = bluelib.find('sensor.lidar.ray_cast')
lidarbp.set_attribute('rotation_frequency', '25')
lidarbp.set_attribute('points_per_second', '100000')

Best regards
Atlunch

Atlunch, 2019-03-13:
found it......Line 633 in manual_control.py

closed 2019-03-13 by Atlunch

vishal-prgmr, 2019-06-05:
Hey. Did you ever try capturing the camera.rgb and the Lidar at the same timestamp using manual_control.py. I need the functionalities of manual_control.py and dont want to write another script from scratch for just serialising the two sensor modalities.