carla-simulator/carla #1376: Help requested: Carla binaries compiled to run with OpenGL 3.

aayushARM, 2019-03-12:
Hello everyone. So I've been really struggling since the last few days to make CarlaUE4.sh launch on Google Colab's VMs(Tesla K80 12GB, Ubuntu 18.04). Colab since my machine doesn't have a dedicated GPU. My plan is to make the server run on Colab VMs and connect to it from local python clients on my machine. The problem is that the latest MESA OpenGL utilities available for Ubuntu 18.04 only support OpenGL 3.x, and the available binaries in CARLA_0.x.x.tar.gz files here are compiled to work with OpenGL 4.3, see [this link](https://github.com/carla-simulator/carla/issues/1253#issuecomment-471838332).
I tried building following the instructions, but the "make" command fails while building Unreal Engine 4, plus I don't even know where to specify -opengl3 flag while building. 

Has anyone compiled it for OpenGL 3, and managed to make CarlaUE4.sh run with -opengl3 flag?
I would really appreciate to have access to those binaries.

I also tried following [this](https://carla.readthedocs.io/en/latest/carla_headless/) guide considering maybe it can help, and everything installs well but I get error on the last step:
```
DISPLAY=:8 vglrun -d :7.0 glxinfo 
name of display: :8
[VGL] ERROR: Could not open display :7.0.
```

Thanks in advance for any help!

aayushARM, 2019-03-12:
Edit: Turns out colab uses a Docker container wherein GPU acceleration is not enabled. Will Try Google Cloud Platform for now.

closed 2019-03-12 by aayushARM

MichaelBosello, 2020-04-20:
I needed to run Carla on Colab too, and I have found a method to do so.
If you're still interested, I did a Colab notebook with the tutorial.
You can find it at https://github.com/MichaelBosello/carla-colab