carla-simulator/carla #1380: How to get  exact pixel in the Town3 from the world coordinates ?  What is the scaling factor ?

imran514, 2019-03-14:
How to get  exact pixel in the Town3 from the world coordinates ?  What is the scaling factor ?

nsubiron, 2019-03-14:
Could you provide more information? pixels of what image are you referring to?

imran514, 2019-03-14:
I want to know the pixel values of where actors(pedestrians/vehicles) are drawn in pygame window

nsubiron, 2019-03-14:
Basically you need to build the intrinsic matrix from FOV and resolution of the image (you have an example here to [convert depth images to 3D points](https://github.com/carla-simulator/carla/blob/0d5a6f335c6293f481ac99bd8a137897da48a604/Deprecated/PythonClient/carla/image_converter.py#L110)), then convert to world coordinates using the camera transform.

nsubiron, 2019-03-14:
No sorry, that's to convert from pixel to 3D point, but doing the inverse should be trivial right? :smile:

nsubiron, 2019-03-14:
Oh wait, check out #1383.

closed 2019-05-20 by stale[bot]