carla-simulator/carla #1386: Is there a way to load map created in VTD(Virtual Test Drive) in CARLA?

77hanzzi, 2019-03-14:
I have created a map in VTD (Virtual Test Drive) and I wanted to load it in CARLA 0.9. I have read the tutorial saying that roadrunner can be used to create map for CARLA by importing .xodr and .fbx files, and similarly my map created in VTD is .xodr file, so I am wondering is there a way to import the map I created in VTD? Thanks.

Sakrust, 2019-03-15:
You can import your .xodr in RoadRunner, then export a  .fbx and a .xodr and import these two in Carla.

77hanzzi, 2019-03-17:
> You can import your .xodr in RoadRunner, then export a .fbx and a .xodr and import these two in Carla.

Thanks for the answer! But the question is I am not able to use this software due to tight budget ...

closed 2019-05-23 by stale[bot]

Lu7913, 2020-07-22:
@HanzhiC Have you figured something out, regarding VTD and Caral or some other tool besides RoadRunner?