carla-simulator/carla #1387: Make Package error: Cook failure

ravishk1, 2019-03-14:
I am using Carla 0.9.3. I have built the Unreal engine successfully, but got an error with the command of 
**make package**  in Ubuntu 16.0.4.

::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
ERROR: Cook failed.
       (see /home/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:13: recipe for target 'package' failed
make: *** [package] Error 25
:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::

My system specs are -
OS-Ubuntu 16.0.4
Graphic card- Nvidia GTX 1080 Ti
i7 processor
RAM 16GB

Any help to find the solution for this will be much appreciated.
Thanks !!!

nsubiron, 2019-03-18:
Hi @ravishk1, could you provide the full log?

Narendarselva, 2019-03-21:
I m using UnreaL Engine 4.21 and carla 0.9.4 
I m getting cook failed error on making package using "make Package" command.
Please find the log attached .
Kinldy help me on this i m stuck for 3 days .

[errorlog.txt](https://github.com/carla-simulator/carla/files/2993097/errorlog.txt)

ravishk1, 2019-03-25:
Hello @nsubiron , please find the full log in the attachment. I'm using Unreal Engine 4.21 and Carla 0.9.3. If possible, please suggest any solution 
[Log.txt](https://github.com/carla-simulator/carla/files/3001753/Log.txt)

nsubiron, 2019-03-26:
@ravishk1 @Narendarselva Have you run `./Update.sh` to retrieve the latest content? It could be some assets are outdated.

closed 2019-06-01 by stale[bot]

chris1234567897, 2021-08-05:
Try running unreal engine as administrator