carla-simulator/carla #1390: how to add the full map?

HeYDwane3, 2019-03-15:
Ubuntu 18.04
Carla 0.9.4

I created a map in roadrunner, but when I exported it and imported into the Carla, I can only see the part within -300<x<300, -300<y<300. Is there any way I can see the full map? I am not sure if it is the problem from Roadrunner or the Carla.

Anyone can help me?

XGodina, 2019-03-18:
Hi, @HeYDwane3 

The problem is from RoadRunner, if you would create a big map, you need to change the world settings. Wich version of RoadRunner do you use?

eloy96vg, 2019-04-25:
Hi @XGodina 

I have the same problem to add the full map. I'm using version 2019.0.4 of RoadRunner. What do i need to change in the world settings to get the full map?

Thanks in advance.

XGodina, 2019-04-26:
Hi @eloy96vg 

If you create a huge map you need to change the world setting.  Move the blue frame until the map stays inside. ( put the mouse above of frame and move it) After that confirm the changes in "Apply World Changes".

Every version RoadRunner change a bit, but the philosophy is the same.

[image]

eloy96vg, 2019-04-27:
Thanks @XGodina , i have solved that problem, but i have tried to generate the map with generate_map.py and the map is much smaller than routeplanners, spawn points, etc.
Do you know why happen this and how solve this problem?

[image]
[image]

Thanks again.

Derekabc, 2019-07-03:
@XGodina Where can I found the Roadrunner files of Town 02 and Town01, so I can make some modification to create a new map.

XGodina, 2019-07-04:
Hi, @Derekabc well, if you download our .xodr files you can load in RoadRunner. RoadRunner should generate the map with OpenDrive information.

closed 2019-09-09 by stale[bot]