carla-simulator/carla #1392: Recording the simulation using camera collected images

krdindorkar19, 2019-03-15:
Hi,

1. I want to record the simulation, for that I have set the recording parameter as True of Camera Manager class (in manual_control.py) and using "ffmpeg" I am trying to create video from the frames and the frames are saved with the frame_number.png but I found that some of the frame numbers are missing. Hence complete video is not generating, I tried it using thread also : 
camera_thread = Thread(target=self.record,args=(weak_self,))
camera_thread.start()
world.thread_list.append(camera_thread)

def record(self, weak_self):
        self.sensor.listen(lambda image: CameraManager._parse_image(weak_self, image))

def _parse_image(weak_self, image):
        if self._recording:
                  image.save_to_disk('camera_out/%08d.png' % image.frame_number, captured_images)

However, this also skipping some frames. 

2. I tried printing timestamp : print(image.timestamp)
but it is giving error, how to know the timestamp of the collected frame

3. Is there is any other way to record the simulation video?

nsubiron, 2019-03-15:
Hi @krdindorkar19,

If the client is too slow parsing the images some frames may drop. You should use synchronous mode for this, check out our example [synchronous_mode.py](https://github.com/carla-simulator/carla/blob/master/PythonAPI/synchronous_mode.py).

krdindorkar19, 2019-03-28:
Hi @nsubiron , Thank you :)
When we do event.save_to_disk(), it saves image with size > 1 mb
Is there is any way to compress this size

nsubiron, 2019-03-29:
You can save as jpeg, that should reduce the size, but don't use it with ground truth data, semantic segmentation and depth will be corrupted if the compression has losses.

closed 2019-04-01 by krdindorkar19