carla-simulator/carla #1397: Big GPU memory usage

pawel-ziecina, 2019-03-15:
Hi,
I noticed that GPU memory usage for 0.9.4 is far bigger comparing to 0.8.4 version. `nvidia-smi` show increase from 1.2GB (can be decreased to half of that with some set of parameters) to 2.75GB.

What is the cause of such big increase? Is it possible to decrease memory footprint?

Best regards
Pawel

nsubiron, 2019-03-15:
Hi @pawel-ziecina,

I haven't measured, but one probable cause is that we disabled texture streaming because it doesn't work with our cameras, only loads high quality textures when the spectator camera is near the object. In 0.8.x it was working cause the spectator camera was a attached to the player, with multi-agent support, there is no player anymore.

It's possible to enable texture streaming in project settings in the Unreal Editor.

pawel-ziecina, 2019-03-15:
Thank you for advice - I'll certainly check it. 

As far as I understand enabled texture streaming will not be an issue in non-rendering mode, right?

nsubiron, 2019-03-15:
No, not at all.

pawel-ziecina, 2019-04-11:
Checked and confirm that switching on `r.TextureStreaming` flag in `Unreal/CarlaUE4/Config/DefaultEngine.ini` helped - GPU memory usage are comparable to those from carla 0.8.4. Tested on carla 0.9.5.

Thanks.

closed 2019-04-11 by pawel-ziecina