carla-simulator/carla #1398: Conversion uasset to uasset/uexp

FrancoisMasson1990, 2019-03-15:
I would like to save Blueprint .uasset from Unreal Editor to .uasset and .uexp in order to run it in LinuxNoEditor. Does someone know how to do it ? 
When replacing the file in Linux Editor I got the following error "Package contains EditorOnly data"

dnovillo-CVC, 2019-03-22:
In other to use a file from the editor to a LinuxNoEditor dist, you need to cook it. Usually, the process of cooking is linked to a map, but you can change that toying around the package settings under project settings inside unreal. Also, make sure that, when you move the blueprint, you also take care of any needed dependency for the blueprint you are trying to add.

closed 2019-05-28 by stale[bot]

guygo12345, 2020-08-09:
Can you explain in more detail how to do this please? @dnovillo-CVC 
My purpose is to edit Vehicle blueprint with UE editor and replace this blueprint in the no-editor version.

Thank you!