carla-simulator/carla #1399: terminate called after throwing an instance of 'std::bad_weak_ptr'

zlw21gxy, 2019-03-16:
[image]

hey guys, I am using Carla 0.9.4 and Ubuntu 16.04
does anyone why this error happen? thanks!!!!
I guess it's due to listening to the sensor in the wrong way

nsubiron, 2019-03-16:
Hi @zlw21gxy, do you have some minimal code example reproducing this issue?

zlw21gxy, 2019-03-28:
thanks for help and sorry for late reply, this days I am trying to locate the error
[image]
it is caused by spawn camera in world, but this error won't happened at the beginning, it only happened when I running program for a while, like couple of hours. @nsubiron

zlw21gxy, 2019-03-28:
#1460

fpasch, 2019-03-28:
I'm facing a similar issue when using the ros-bridge (and starting/stopping another client).
I believe the issue is related to having two clients accessing the data.

My clients:
1. ros-bridge
2. manual_control.py

You can reproduce it like that.

1. execute ros-bridge
2. Restart manual_control.py every 4 seconds (e.g with the help of this script)
```
#!/usr/bin/env python
import signal
import subprocess
import time

ret = 0
while ret is 0:
    pro = subprocess.Popen("./manual_control.py")
    time.sleep(4)
    pro.terminate()
    pro.wait()
    ret = pro.returncode
print("returned {}".format(ret))
```

Afer a few restarts of manual-control, the ROS bridge will report the error and stops.

zdx3578, 2019-03-29:
@nsubiron

nsubiron, 2019-03-29:
It looks like it's a bug inside LibCarla, related to the client objects used to subscribe to a sensor. I'll try to create a minimal test reproducing this in C++. Unfortunately we won't be able to fix it for 0.9.5.

oldtopos, 2019-06-17:
When runnng ros_bridge, I commented line 167 of parent.py

self.child_actors[actor_id].destroy()

This has greatly improved ros_bridge stability; I expect it is causing a double delete attempt in the python API code ...

fpasch, 2019-06-18:
@oldtopos What you're describing is a different issue, which is, as you already stated ros-bridge only. I've create an issue in ros-bridge > https://github.com/carla-simulator/ros-bridge/issues/133

germanros1987, 2020-03-19:
This should be solved now in 0.9.8.

closed 2020-03-19 by germanros1987