carla-simulator/carla #1404: Is it possible to change right-hand traffic rule to left hand?

robinalen, 2019-03-18:
OS: Ubuntu
Version: 0.9.4

Hi everyone,
I wanna change traffic rule to left hand driving when I randomly spawn cars in UE engine. I tried the lane change API in the latest Carla version but it did not work. 
Thanks

nsubiron, 2019-03-18:
Hi @robinalen,

It's not that simple, you could adapt the autopilot to drive following "opposite" direction, but not everything in the city is reversible, you would need to replace every traffic signal, for instance.

robinalen, 2019-03-18:
> Hi @robinalen,
> It's not that simple, you could adapt the autopilot to drive following "opposite" direction, but not everything in the city is reversible, you would need to replace every traffic signal, for instance.

Hi ＠nsubiron, thank you for the replay. Yes, that's what I was thinking previously, just crudely let the car drive in opposite direction and change traffic signals.

Abanoub-G, 2019-04-15:
Hi, 
   I have my own map that I am adding to it my own assets and doing the necessary map coding i.e. modifying the traffic lights to work for left handed systems, speed signs, roundabouts etc.

I was wondering if you could give more details of how to invert the vehicles to drive in the "opposite" direction?  

Thanks

robinalen, 2019-04-16:
Hi @Abanoub-G ,
I haven't go that deep so far, but since you can create your own map why not change the travel direction while map making? 
Like this:
[image]
I'm not sure if this works for Carla, I'll try to make a left-hand map next month.

Abanoub-G, 2019-04-17:
Hi, 
Thanks a lot for your reply. The thing is that I am not building this on road runner. I have a map that was build by a professional unreal graphics artist and I am trying to figure out how to add the paths on it. 

Are these paths (splines) generated using road maps i.e. is it something that road maps create automatically when creating the maps?

Thanks

robinalen, 2019-04-18:
Hi,
Those splines are automatically generated by road runner. I don't have a time to look the details about it but the CarlaDoc seems mentioned a way to modify those splines in UE.
Links: [https://carla.readthedocs.io/en/latest/map_customization/](url)

nsubiron, 2019-04-18:
It's not documented, but you can make the autopilot work by adding _RoutePlanners_ all over the map. Now, if you have an OpenDrive file associated with your map you can automatically generate these route planners with the _OpenDriveActor_. If not you'll have to it manually, drag and drop a _Route Planner_ from the tab into the map
[image]
[image]

Add more spline points by dragging a point while pressing "alt".

Abanoub-G, 2019-04-25:
Thanks that certainly helped.

closed 2019-07-01 by stale[bot]

bapukokare, 2022-11-25:
What is the way directly interchange road traverse direction? condition is  to use open drive files 
[lht_trial.txt](https://github.com/carla-simulator/carla/files/10089472/lht_trial.txt)
with notepad only

ivevasiljevic, 2023-03-22:
Any updates on this matter, with the version 0.9.13? On my map I have a single right hand side road and would like to make the car go into the opposite direction without making a new identical map with a single left hand side road.

So is it possible to accomplish this through code? Or editing some xodr files?

JIANGZhengm, 2024-09-11:
> Any updates on this matter, with the version 0.9.13? On my map I have a single right hand side road and would like to make the car go into the opposite direction without making a new identical map [...]

> Any updates on this matter, with the version 0.9.13? On my map I have a single right hand side road and would like to make the car go into the opposite direction without making a new identical map [...]

I have the sampe issue, have you resolved this problem?