carla-simulator/carla #1406: Engine crash handling finished; re-raising signal 11 for the default handler

zlw21gxy, 2019-03-18:
Hey guys, I am using the latest Nvidia driver but still can't use Carla normally...does anyone face the same problem? thanks
[image]
[image]

zlw21gxy, 2019-03-18:
I am using Ubuntu 16.04 LTS.....

zlw21gxy, 2019-03-22:
when download nvidia driver version to NVIDIA-SMI 418.39       Driver Version: 418.39  it is ok;
sudo apt-get install nvidia-418 install the stable version ;  @CreateAMind.ai

closed 2019-05-28 by stale[bot]

githubwys, 2020-09-23:
can someone deal with this problem?
i also meet this problem and need help, thank you.