carla-simulator/carla #1412: Loading multiple maps in sequence

felipecode, 2019-03-19:
The following code excerpt fails:

```
import carla
client = carla.Client('localhost', int(2000))
client.set_timeout(25.0)
world = client.load_world('Town01')
settings = world.get_settings()
settings.synchronous_mode = True
world.apply_settings(settings)

world = client.load_world('Town02')
```

Basically it seems that you need to call world.tick() before loading another
map. I think map loading should not need to wait for client synchronization.

fedor-chervinskii, 2019-04-01:
Hi @felipecode, I've just experienced the same behavior, and it turns out that calling a tick is not enough in my case. The only workaround I found is to switch synchronous mode off and then after loading a map switch it back on.

So basically the issue here is that re-loading a map is not working in synchronous mode.

felipecode, 2019-04-01:
Yep, Just tick is not working. And if the client disconect without removing the synch mode you cannot reconect.

pawel-ziecina, 2019-04-02:
alternatively you can send "asynchronously" request for tick to server, ex:

```python
threading.Timer(1., world.tick).start()
world = client.load_world('Town02')
```

But its rather temporary fix :)

nsubiron, 2019-04-29:
It seems we forgot to reset the settings after a new world is loaded, the intended behaviour was to reload the map with default settings (because a fresh new map is created), but they were not properly notified to the engine.

The following works now (also without ticking in between).
```py
client = carla.Client('localhost', 2000)
client.set_timeout(2.0)

world = client.get_world()

while True:
    print('new world: enable sync mode.')
    world.apply_settings(carla.WorldSettings(synchronous_mode=True))

    for _ in range(0, 10):
        time.sleep(1)
        world.tick()
        world.wait_for_tick()
        print('tick!')

    world = client.reload_world()
```

PR #1588 (2019-05-10): Fix dead-lock when loading a new map in synchronous mode.
#### Description

It seems we forgot to reset the settings after a new world is loaded, the intended behaviour was to reload the map with default settings because a fresh new map is created, and in fact the new world had the settings reloaded but the engine was not properly notified thus creating inconsistent state.

The following works now (also without ticking in between).
```py
client = carla.Client('localhost', 2000)
client.set_timeout(2.0)

world = client.get_world()

while True:
    print('new world: enable sync mode.')
    world. [... cut]
commit cf58f7a2 (2019-05-10): Fix dead-lock when loading a new map in synchronous mode. (#1588)

* Fix #1412, dead-lock when loading a new map in synchronous mode.
* Update CHANGELOG
* Add smoke test for #1412
closed 2019-05-10 by nsubiron