carla-simulator/carla #1420: WheelPhysicsControl content misleading/invalid

fpasch, 2019-03-20:
I see some issues with WheelPhysicsControl:

**1. Missleading name**
The WheelPhysicsControl has a member ```steer_angle```.
This is somehow misleading, as it's not the current angle but most likely the max angle.
To clarify things, we should rename it to ```max_steer_angle```.

**2. disable_steering always false**
I see that for my vehicle the member ```disable_steering``` is always false.
I would assume, that this member shows which wheel is fixed (disable_steering=True), and which one can turn (disable_steering=False).
Also, maybe we can get rid of the negation by using ```is_steerable```.

fpasch, 2019-03-20:
It would also make sense to be able to identify each wheel. Currently VehiclePhysicsControl only contains an array of wheels.

manishthani, 2019-04-24:
Hi @fpasch,
Just checked again and indeed you are right!

 I've included the corresponding changes about renaming those 2 parameters in branch #1562 and made a PR already.

Thank you very much for your feedback. Really appreciate it!

closed 2019-05-29 by nsubiron