carla-simulator/carla #1422: Force feedback, wheel centering and steering wheel range using Logitech Wheel

mallela, 2019-03-20:
I had some questions regarding the driving using the Logitech wheel. 

I am trying out the `maual_control_steeringwheel.py` code from CARLA 0.9.4 and I noticed that the recovery from a sharp turn is difficult and is in fact easier using keyboard controls. Is this due to the range of the steering wheel? Where can I change it? 

Also, how do I enable force feedback? It does not come readily with the kernel. It would also help to have the wheel centering property for the wheel. How can I get this?

damascenodiego, 2019-05-03:
@mallela 

I managed to enable force feedback by using the [python-evdev](https://github.com/gvalkov/python-evdev) library. In the Logitech G29, there is a parameter named FF_AUTOCENTER that you can set values to set levels to the autocenter.

[The code to test the autocenter feature is here](https://gist.github.com/damascenodiego/7fcac91e44033a655f8c89ba2aebfd9a)

closed 2019-07-09 by stale[bot]

LeehamElectronics, 2021-04-16:
Does anyone have any examples of a python script that interacts with steering wheel force feedback? I can't seem to figure out how to control force feedback on steering wheels.
Thanks

quangsonle, 2021-08-08:
> @mallela
>
> I managed to enable force feedback by using the [python-evdev](https://github.com/gvalkov/python-evdev) library. In the Logitech G29, there is a parameter named FF_AUTOCENTER that you can [...]
dear,
As evdev is only for Linux, do you have any idea to make it work on Windows?

josealgruiz, 2023-01-13:
@quangsonle Hello!

Where you able to solve this issue for windows? 
Since as you mentioned evdev is only for Linux.

Thanks :)

quangsonle, 2023-01-13:
@josealgruiz i solved it by a communication between c# and python. It has been more than a year, there should be alternative solutions

josealgruiz, 2023-01-13:
@quangsonle 

Wow! Thanks for the quick answer wasn't expecting it :D 

Would you mind sharing your solution? or give some hints/ideas? I'm stuck in this and I need it for research purposes :(

quangsonle, 2023-01-19:
@josealgruiz, my solution: control the wheel on c# (read wheel and pedal information, set constant feedback), which is done easily on window with Logitech library, sending wheel information to python that runs carla client.
I only have the code at c#'s side and the sending one, the python code is modified from carla's manual driving with steering wheel (just the steering wheel data is read from c#)

Marcosciaco, 2024-11-05:
Could you maybe share the C# and python code? I am working on the same thing and have no real clue where to start

quangsonle, 2024-11-05:
@Marcosciaco let's break it down and do it step-by-step:
1. The wheel side: ensure you have the Logitech driver installed, you have the SDK, and can run the example in the SDK (to read the values of the wheel: you should see an int value changed when you revolve the wheel) - you have done this step.
2. Python client side of Carla: there is the manual driving.py after you successfully install everything. Ensure you can run it and control the car with your keyboard - you have done this step.
3. Combine them: there are many ways to send the values from c# to Python, one is on my repository, and you Can try it.
Note: there is a way to use the Logitech wheel directly on Python but you cannot manipulate the force feedback as I mentioned. Only through c#, we can control the force feedback as far as I know