carla-simulator/carla #1423: recipe for target 'LibCarla.server' failed

srinikrish22, 2019-03-21:
Hi community, 

I have been trying to compile on Linux since a week but in vain. I have already faced issues related to issue #742 which was solved by installing boost_1.67 manually from source. But now I face a new issue when making launch :

[12/43] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o
FAILED: /usr/bin/clang++-6.0   -DASIO_NO_EXCEPTIONS -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../boost-1.69.0-install/include -I../rpclib-v2.2.1_c1-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -stdlib=libc++ -I/media/<username>/<partition>/carla/carla/Build/llvm-6.0-ex-install/include/c++/v1 -fno-exceptions -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -stdlib=libc++ -I/media/<username>/<partition>/carla/carla/Build/llvm-6.0-ex-install/include/c++/v1 -fno-exceptions   -O3 -DNDEBUG -MMD -MT LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o -c ../../LibCarla/source/carla/road/element/LaneCrossingCalculator.cpp
../../LibCarla/source/carla/road/element/**LaneCrossingCalculator.cpp:37:39: error: no member named 'has_value' in 'boost::optional<carla::road::element::Waypoint>'
    return !map.GetWaypoint(location).has_value();**
            ~~~~~~~~~~~~~~~~~~~~~~~~~ ^
1 error generated.
[12/43] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o
FAILED: /usr/bin/clang++-6.0   -DASIO_NO_EXCEPTIONS -DBOOST_ASIO_ENABLE_BUFFER_DEBUGGING -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_NO_EXCEPTIONS -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true -DLIBCARLA_NO_EXCEPTIONS -DPUGIXML_NO_EXCEPTIONS -I../../LibCarla/cmake/../source -I../boost-1.69.0-install/include -I../rpclib-v2.2.1_c1-libcxx-install/include -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -stdlib=libc++ -I/media/<username>/<partition>/carla/carla/Build/llvm-6.0-ex-install/include/c++/v1 -fno-exceptions -std=c++14 -pthread -fPIC -Werror -Wall -Wextra -stdlib=libc++ -I/media/<username>/<partition>/carla/carla/Build/llvm-6.0-ex-install/include/c++/v1 -fno-exceptions   -g -MMD -MT LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o -MF LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o.d -o LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/road/element/LaneCrossingCalculator.cpp.o -c ../../LibCarla/source/carla/road/element/LaneCrossingCalculator.cpp
../../LibCarla/source/carla/road/element/LaneCrossingCalculator.cpp:37:39: error: no member named 'has_value' in 'boost::optional<carla::road::element::Waypoint>'
    return !map.GetWaypoint(location).has_value();
            ~~~~~~~~~~~~~~~~~~~~~~~~~ ^
1 error generated.
[12/43] Building CXX object LibCarla/cmake/server/CMakeFiles/carla_server_debug.dir/__/__/source/carla/streaming/detail/tcp/Client.cpp.o
ninja: build stopped: subcommand failed.
Util/BuildTools/Linux.mk:80: recipe for target 'LibCarla.server' failed
make: *** [LibCarla.server] Error 1

Hoping to get some help from the community. I am on Ubuntu 16.04 with anaconda python 3.7

closed 2019-05-27 by stale[bot]

Stan-lee-1229, 2019-09-14:
same how to solve it！

zkyxsq, 2020-03-03:
same how to solve it！

Charan-Karthikeyan, 2020-07-24:
Has anyone found the solution to the error. I have tried to remove and reinstall the carla module. But always end up with this error. 
If anyone knows what to do, please do share.