carla-simulator/carla #1424: Unexpected behavior at vehicle initialization

germanros1987, 2019-03-21:
* CARLA version(s): 0.9.4 , overnight 2019-03-19 and probably others
* Platform: Ubuntu 16.04
-----

When spawning a vehicle, some of the calls are asynchronous and get_location (and probably other methods) return wrong values for a while.

Code to reproduce the issue:
```
import carla
import time

client = carla.Client('localhost', 2000)
client.set_timeout(10)
world = client.load_world('Town04')

blueprint_library = world.get_blueprint_library()
blueprint = blueprint_library.filter('vehicle.lincoln.mkz2017')[0]
actor = world.try_spawn_actor(blueprint, carla.Transform(carla.Location(131, -182, 0.1)))

loc = actor.get_location()
print(loc)
time.sleep(2)
loc2 = actor.get_location()
print(loc2)
```

Output:

```
Location(x=0, y=0, z=0)
Location(x=131, y=-182, z=0.0295535)
```

fabianoboril, 2019-03-21:
I observed that also in an older version of CARLA. The solution was to put a world.wait_for_tick() right after the load command.

nsubiron, 2019-03-21:
This is a consequence of the optimization we do: `get_transform` instead of calling the simulator returns the transform that was received last tick. If we ask for the transform right after spawning an actor, this actor was not yet present in the last tick and it returns a defaulted actor state. The system is explained [here](https://github.com/carla-simulator/carla/issues/1274#issuecomment-465567495). Also related #1151, we found a workaround for this one, but it's giving some issues with synchronous mode.

It is difficult to solve this internally without affecting performance.

But as @fabianoboril mentioned, doing a `wait_for_tick` after spawning actors is probably the best workaround.

```py
actor0 = world.spawn_actor(...)
actor1 = world.spawn_actor(...)
actor2 = world.spawn_actor(...)
world.wait_for_tick()
transform = actor0.get_transform()
```

germanros1987, 2019-03-21:
Some additional tests.

Running this code snippet:

```
import carla
import time

client = carla.Client('localhost', 2000)
client.set_timeout(10)
world = client.load_world('Town04')
world.wait_for_tick()

blueprint_library = world.get_blueprint_library()
blueprint = blueprint_library.filter('vehicle.lincoln.mkz2017')[0]
actor = world.try_spawn_actor(blueprint, carla.Transform(carla.Location(131, -182, 0.1)))

loc = actor.get_location()
print(loc)
time.sleep(2)
loc2 = actor.get_location()
print(loc2)

```

I get:

```
Location(x=0, y=0, z=0)
Location(x=131, y=-182, z=0.0295535)

```

Running this one:

```
import carla
import time

client = carla.Client('localhost', 2000)
client.set_timeout(10)
world = client.load_world('Town04')
world.wait_for_tick()

blueprint_library = world.get_blueprint_library()
blueprint = blueprint_library.filter('vehicle.lincoln.mkz2017')[0]
actor = world.try_spawn_actor(blueprint, carla.Transform(carla.Location(131, -182, 0.1)))
world.wait_for_tick()
loc = actor.get_location()
print(loc)
time.sleep(2)
loc2 = actor.get_location()
print(loc2)

```

I get

```
Location(x=131, y=-182, z=0.1)
Location(x=131, y=-182, z=0.0295535)

```

So indeed it seems we need a wait_for_tick after the spawning of a set of agents.

closed 2019-05-28 by stale[bot]

yaodongC, 2023-02-15:
if you are in synchronize mode, you should add **world.tick()** right after the try_spawn  method to avoid this problem