carla-simulator/carla #1425: Can Carla simultaneously simulate real-world vehicles?

slowergmajor, 2019-03-21:
hello everyone，

I am conceiving a project to realize the mirror display of the real vehicle on the simulation platform with Carla. I want to know if the current version of Carla supports this. Suppose I can build a road model that is consistent with the real scene.

I find that current Carla can generate vehicles at specified coordinates with"spawn_actor" or"try_spawn_actor", and there are some scripts in agents folder to achieve simple navigate from point A to point B use a PID controller to calculate the throttle and steer. 

If I can only get the basic status information of the real vehicle, such as the location，speed and direction. Obviously, it is very difficult to keep the dynamic model of the simulated vehicle completely consistent with the specific real vehicle. In other words, it is difficult to synchronize the simulated vehicle with the real vehicle just by calculating the throttle and steer. It also needs constant position correction to keep up with the real position. 

So when the simulation position deviates from the real position, what method can be used to smoothly calibrate the position of the vehicle in Carla?

I don't know if my expression is clear, I hope to discuss with your genius.

nsubiron, 2019-03-22:
Hi @slowergmajor,

You can use `set_transform`, that teleports the vehicle to the desired location and rotation. You can also set velocity/angular velocity if needed. Would that work for your case?

slowergmajor, 2019-03-22:
> Hi @slowergmajor,
> 
> You can use `set_transform`, that teleports the vehicle to the desired location and rotation. You can also set velocity/angular velocity if needed. Would that work for your case?

Thank you for your reply! I tried `set_transform` and it seems to solve my position correction problem.
 
And as you say I can set velocity/ angular velocity. Do you mean that I can set the velocity/ angular velocity of the vehicle directly without setting the throttle? May I ask which specific command can achieve this function? I noticed that there is a `carla.command.ApplyVelocity` in the module of your documentation. But I tried it and didn't find out how to use it. Do you have any examples that use this feature?

Thank you.

nsubiron, 2019-03-22:
Set velocity and angular velocity are applied directly to the physics actor, do not take into account the vehicle dynamics, they aren't a substitute of `VehicleControl`. For instance, the following code will make the vehicle jump several meters over the ground
```py
vehicle.set_velocity(carla.Vector3D(z=15))
```

The objects inside `carla.command` module are commands to use with the `client.apply_batch` function. They're used to apply a certain command to a lot of actor simultaneously, e.g. making all the vehicles jump
```py
vehicles = world.get_actors().filter('vehicle.*')
client.apply_batch([carla.command.ApplyVelocity(x.id, carla.Vector3D(z=5)) for x in vehicles])
```
They're there mostly for performance reasons, as it's way more efficient sending a single message to the simulator than a message per actor.

slowergmajor, 2019-03-23:
> Set velocity and angular velocity are applied directly to the physics actor, do not take into account the vehicle dynamics, they aren't a substitute of `VehicleControl`. For instance, the following code will make the vehicle jump several meters over the ground
> 
> ```python
> vehicle.set_velocity(carla.Vector3D(z=15))
> ```
> 
> The objects inside `carla.command` module are commands to use with the `client.apply_batch` function. They're used to apply a certain command to a lot of actor simultaneously, e.g. making all the vehicles jump
> 
> ```python
> vehicles = world.get_actors().filter('vehicle.*')
> client.apply_batch([carla.command.ApplyVelocity(x.id, carla.Vector3D(z=5)) for x in vehicles])
> ```
> 
> They're there mostly for performance reasons, as it's way more efficient sending a single message to the simulator than a message per actor.

Thank you for your example. I tried use your example "throw the car", and it was really funny. At the same time, I also tried setting the angular velocity `vehicle.set_angular_velocity(carla.Vector3D(z=1000))`
it can make the vehicle spin around.
So this is to give an instantaneous speed without affecting the basic physical characteristics of the UE engine. For example, here I tried to give the vehicle an instantaneous longitudinal speed, and the vehicle will slowly stop, instead of keeping this speed going.In the same way, the control of the angular acceleration of the vehicle is more elusive due to the existence of the physical model. This is also the reason why I set z to 1000 above. When set to 100, the vehicle turned almost a little bit slightly. So if I want to demonstrate a fixed angular acceleration, I am afraid it is difficult to accurately implement this command.

But thank you very much for your answer, it is still very helpful to me.

I also want to ask, I don't know if there is any clearer source code for these command? I found that there are some features that are not listed on your webpage, such as `set_velocity``set_angular_velocity``get_angular_velocity`and so on.If you can tell me the source location of these functions, we can more easily explore the functions of Carla ourselves.

Thanks anyway!

nsubiron, 2019-03-26:
Oh true, we're missing some functions, thanks I'll add them.

The Python bindings are all inside _"PythonAPI/source/libcarla"_, e.g. [binding for actors](https://github.com/carla-simulator/carla/blob/3645299e29c7b24c41c0a334505a64a881e863d8/PythonAPI/source/libcarla/Actor.cpp#L56-L84). All these binding call C++ functions inside LibCarla, e.g. [Actor.h](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/client/Actor.h).

closed 2019-06-01 by stale[bot]