carla-simulator/carla #1427: How does the function 'vehicle.apply_control' actually work?

IreneLu12, 2019-03-22:
Hi,
I am trying to figure out the vehicle dynamic model of CARLA. And I know we can apply series of control command like 'throttle, steer..'. But how does these parameters been actually used? Is there any formula we can refer to ? 
Thanks!

togaen, 2019-03-28:
I am also interested in this information (and I think #1316 is asking a similar question). Along these lines, valid vehicle control values do not seem to be documented anywhere. According to @nsubiron, they are: throttle and brake [0, 1], steer [-1, 1].

nsubiron, 2019-03-29:
Hi,

Sorry about the lack of documentation. The values in VehicleControl struct are directly passed to the PhysX Vehicle plugin (there is some extra logic on top for the gear change, see [CarlaWheeledVehicle.cpp](https://github.com/carla-simulator/carla/blob/422d0de1c4ba34238d960a5405ab227044199e13/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp#L78-L102)).

closed 2019-06-04 by stale[bot]

tsc1989, 2019-10-16:
All values are finally pass to the vehicle PID controller. 
have a look in to file **PythonAPI/carla/agents/navigation/controller.py** 
and classes: **PIDLongitudinalController**() & **PIDLateralController**()

For speed control it uses Longitudinal controller.

The overall PID control function can be expressed mathematically as described in wiki page --> **Mathematical form**

> https://en.wikipedia.org/wiki/PID_controller

Python classes does the same thing here, it derives derivative & integral part from current & target speed and finds final u(t) value. which will be applied to vehicleControl

same applies to steering control also (PIDLateralController)