carla-simulator/carla #1428: make launch/package fails

Narendarselva, 2019-03-22:
I m using UnreaL Engine 4.21 and carla 0.9.4
I m getting cook failed error on making package using "make Package" command.
Please find the log attached .
Kinldy help me on this i m stuck for 3 days .
[errorlog.txt](https://github.com/carla-simulator/carla/files/2995295/errorlog.txt)

nsubiron, 2019-03-22:
Hi @Narendarselva,

Did you download the asset with `Update.sh`?

Narendarselva, 2019-03-22:
Thanks with the update the issue got fixed

closed 2019-03-22 by Narendarselva

kaushikb258, 2019-06-04:
Hello, I'm getting the same error. I'm using Ubuntu 16.04, Carla 0.9.5 and Unreal Engine 4.22. I followed the same instructions as given in the Carla website: 1) ./Update.sh, 2) make launch, 3) make PythonAPI, 4) make package. It failed in the make package step. Also, all this is in the master branch (on stable branch, it failed to make launch.) Any help welcome...