carla-simulator/carla #1431: Spawning new vehicle asset using Python API

pparmesh, 2019-03-24:
I wanted to use a new vehicle asset which I bought from Epic Games Marketplace (https://www.unrealengine.com/marketplace/en-US/truck-trailer-attach-detach-system). I created a project in Unreal Engine using this vehicle asset and then migrated it to my CARLA project. However, I am not able understand how do I spawn it using the Python API.  I have referred the page on how to add assets (https://carla.readthedocs.io/en/latest/how_to_add_assets/) but I think the link explains how to create blueprints etc for my vehicle which I already have. 
Can someone please explain how do I add this vehicle into my CARLA project as well as control it?
I am using Ubuntu 16.04 LTS

kimiya66, 2019-03-26:
spawn_actor(blueprint, spwan_point) is not working for you?

pparmesh, 2019-03-28:
@kimiya66 So I opened the project that had my truck in it and "migrated" it to the content folder of Carla. Can you please tell me what to do next? Basically I want to manually control the truck as well as using a script. Thank you so much!

closed 2019-06-03 by stale[bot]

Akshay-sudo-ux, 2020-03-13:
@pparmesh Were you able to spawn that Vehicle?