carla-simulator/carla #1432: Linking errors when compiling CarlaUE4 from source in Visual Studio

Yougmark, 2019-03-24:
``make launch`` fails to launch the editor.  It says CarlaUE4 and Carla modules are missing or built with different engine version.  But I have only 4.21 version of UE in my system.

Then I chose to rebuild them.  It fails rebuilding CarlaUE4.  It's suggested to rebuild from source.  Then I generated Visual Studio project and tried compiling it in Visual Studio.  There are linking errors shown below.  Any suggestions/help to solve these errors?

I'm using the up-to-date (03-23-2019) master branch of Carla (v0.9.4) , UE v4.21, Visual Studio 2017 on Windows 10.  Has anyone succeeded with this version of Carla?  Suggestions on any working version?

Thanks!

[image]
[image]
[image]

nsubiron, 2019-03-25:
Hi @Yougmark,

Try with the latest release tag, 0.9.4, that should compile

    git checkout 0.9.4

Yougmark, 2019-03-28:
The 0.9.4 release compiles.

closed 2019-03-28 by Yougmark