carla-simulator/carla #1434: make launch error

xiaoyang666, 2019-03-24:
Setup.sh: llvm-6.0-ex already installed.
Setup.sh: boost-1.69.0 already installed.
Setup.sh: rpclib-v2.2.1_c1 already installed.
Setup.sh: googletest-1.8.0-ex already installed.
Setup.sh: CARLA version 0.8.4-dirty.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Server.Release" configuration.
ninja: no work to do.
[1/1] Install the project...
-- Install configuration: "Server"
BuildLibCarla.sh: Success!
BuildCarlaUE4.sh: ERROR: UE4_ROOT is not defined, or points to a non-existant directory, please set this environment variable.
Util/BuildTools/Linux.mk:7: recipe for target 'launch' failed
make: *** [launch] Error 2

This problem has been bothering me for a long time.Please help me . Thanks.

SpakMiguel, 2019-03-25:
Have you skipped the setting UE4 variable step?

You should type the following before "make launch":

_export UE4_ROOT=~/UnrealEngine_4.21_

where _~/UnrealEngine_4.21_ is the absolute path to the installation directory of UE4 in your PC.

xiaoyang666, 2019-03-25:
I've solved this problem. Thank you.
export UE4_ROOT=~/UnrealEngine_4.21 This file is best placed in the Homo directory

closed 2019-05-31 by stale[bot]