carla-simulator/carla #1436: Solving NotImplementedError from PythonAPI/setup.py on Linux Mint

KMint1819, 2019-03-25:
Carla:0.9.4
os:LinuxMint 18.04

For those who get the following errors while building PythonAPI on Linux Mint:
`Traceback (most recent call last):
  File "setup.py", line 114, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 57, in get_libcarla_extensions
    raise NotImplementedError
NotImplementedError
Util/BuildTools/Linux.mk:74: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1
`
Go to (Carla_ROOT)/PythonAPI/setup.py and add "linuxmint" on line 32, and it wil be all fine.

closed 2019-05-31 by stale[bot]

qhaas, 2020-06-24:
Got the same error on EL7, adding `"centos", "redhat"` allowed me to proceed, thanks!

philippwulff, 2020-10-22:
I get the same error message. However writing "linuxmint" in there does not solve the error for me.

philippwulff, 2020-10-22:
I fixed my issue.

The source code in setup.py throw a `NotImplementedException` if 
`distro.linux_distribution()[].lower() is not in ["ubuntu", "debian", "deepin"]`

You need to add `"linux mint"` to this list to fix it.

Here is the python console output.
```

Python 2.7.17 (default, Sep 30 2020, 13:38:04) 
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import distro
>>> distro.linux_distribution()
('Linux Mint', '19.3', 'tricia')
>>> distro.linux_distribution()[0].lower()
'linux mint'
```