carla-simulator/carla #1439: pygame in python samples

kimiya66, 2019-03-26:
Hi,
I do not understand why most of the python samples that you provided(like manual_control.py or automatic_control-py) imported pygame and run on pygame window and why they do not just use Unreal. 
I mean we should have 2 graphical windows(pygame and unrealEngine as carla-server) isn't it possible that all projects execute in CarlsUE4.sh or unreal window. Is this because of cameras and sensors or object recognition that just work with pygame. Because it makes more complicated to connect pygame to unreal. could some one explain me why we need pygame in some projects?
Thank you

analog-cbarber, 2019-03-26:
I am not a member of this project, but I believe the point of having a standalone window in a separate process is to mimic how a autonomous driving agent interacts with CARLA. A in-game controller would only be useful for manual control, and that's not really the point of this project.

In any case, you definitely do not need pygame to use CARLA. We do not use it in our own project.

closed 2019-06-01 by stale[bot]

robogast, 2020-01-16:
Is there still a way in which this manual control of a vehicle is able to run natively in the unreal engine?

As it stands, I'm trying to compare human-vs-machine performance in a driving task, and seeing as the performance of the pygame implementation leaves something to be desired, having the human control a vehicle directly in unreal seems the best option.

I wouldn't mind having to implement it myself, but as I have no experience in developing for unreal, let alone unreal + carla, I don't really know where to begin.