carla-simulator/carla #1442: Unable to apply new carla.VehiclePhysicsControl wheels parameters

DanielsJusts, 2019-03-26:
Hello!

I am trying to change the `steer_angle` parameter of a vehicle, but the values are not updated.
I was hoping that someone would be able to point out what I'm doing wrong.

Here is an example of how I'm attempting this:
```
import carla
import random

client = carla.Client(host="127.0.0.1", port=2000)
client.set_timeout(2.0)
world = client.get_world()

bp_library = world.get_blueprint_library()
vehicle_bp = bp_library.find('vehicle.bmw.isetta')

spawn_point = random.choice(world.get_map().get_spawn_points())

vehicle = world.try_spawn_actor(vehicle_bp, spawn_point)

front_wheel = carla.WheelPhysicsControl(steer_angle = 20.0, disable_steering=False, damping_rate=0.25, tire_friction=3.5)
rear_wheel  = carla.WheelPhysicsControl(steer_angle = 0, disable_steering=False, damping_rate=0.25, tire_friction=3.5)
vehicle.apply_physics_control(carla.VehiclePhysicsControl(wheels = [front_wheel, front_wheel, rear_wheel, rear_wheel]))
```

Is this the correct way to do this?
I am using Python 3.5.2. on Ubuntu 16.04 with Carla 0.9.4.

slowergmajor, 2019-03-30:
If you only want to control the vehicle's steer angle, maybe you can try the `apply_control` method to control the vehicle front wheels' steer angle.
like
```
control = vehicle.get_control()
control.steer = 20
vehicle.apply_control(control)
```

DanielsJusts, 2019-04-01:
Hello, @slowergmajor!
What I'm actually trying to accomplish is to change the VehiclePhysicsControl parameter _maximum_ `steer_angle` of the car. 
I edited my post to clarify this, but thank you for your input!

manishthani, 2019-04-24:
Hi @DanielsJusts  and @slowergmajor ,

Sorry for late reply, I have done a Pull Request ( #1562 ) with changes that solves this issue since it seemed to be buggy. In the following days, these changes will be merged in `master` branch and you will be able to use it. I apologize for the delay.

**Furthermore**, please keep in mind that I have also applied changes related to issue #1420 , so basically, `steer_angle` will be renamed as `max_steer_angle` and `disable_steering` will be renamed as `is_steerable` since it was misleading.

Here is a sample code that would work, it is very similar to the one you had:
```py
# Create front and back Wheels Physics Control
front_wheel = carla.WheelPhysicsControl(
    tire_friction=4.5,
    damping_rate=1.0,
    max_steer_angle=70.0,
    is_steerable=True
)

back_wheel = carla.WheelPhysicsControl(
    tire_friction=4.5,
    damping_rate=1.0,
    is_steerable=False
)

wheels = [front_wheel, front_wheel, back_wheel, back_wheel]

# Create Vehicle Physics Control
torque_curve = [carla.Vector2D(x=0, y=400),
                carla.Vector2D(x=1300, y=600)]

steering_curve = [carla.Vector2D(x=0, y=1),
                  carla.Vector2D(x=100, y=1),
                  carla.Vector2D(x=300, y=1)]

physics_control = carla.VehiclePhysicsControl(
    torque_curve=torque_curve,
    max_rpm=10000,
    moi=1.0,
    damping_rate_full_throttle=0.0,
    damping_rate_zero_throttle_clutch_engaged=0,
    damping_rate_zero_throttle_clutch_disengaged=0.0,
    use_gear_autobox=True,
    gear_switch_time=0.5,
    clutch_strength=10,
    mass=10000,
    drag_coefficient=0.25,
    center_of_mass=carla.Vector3D(x=0, y=0, z=-10),
    steering_curve=steering_curve,
    wheels=wheels
)

# Apply Vehicle Physics Control for a given carla.Vehicle
vehicle.apply_physics_control(physics_control)
```
Please, let me know if it worked for you and thank you very much for your feedback!

DiegoOGV, 2019-05-08:
Hello @manishthani

I have carla 0.9.4 on windows 10, I'm trying your solution but It doesn't work, but when I comment all the cod releated to the wheel physics it works. I suppose there are some problem with the wheel physics on the version.

BR

DanielsJusts, 2019-05-09:
Hello @DiegoOGV!

As @manishthani stated, he wrote a sample code for a fix that is going to be merged into `master` branch, where the names of `WheelPhysicsControl` attributes will be changed, therefore his code is not compatible with already released versions of Carla (in your case - 0.9.4).

If you read @manishthani comment carefully, you can see that he already states that in the previous version the names of `WheelPhysicsControl` attributes were a little misleading, therefore they will be changed un the upcoming release of Carla.

DiegoOGV, 2019-05-14:
Thanks @DanielsJusts!!

I have another question on this couple of code. I saw if I need to access the center of mass on the x coordinate I need just to change "physics_control.center_of_mass.x", hence I will change the value of this variable. 
I'm trying to do the same for to the torque curve, so 

a = carla.Vector2D(0.0,450.0)
b = carla.Vector2D(x = 5000.0, y = 350.0)
torque_curve = [a,b]

physics_control = carla.VehiclePhysicsControl(torque_curve=torque_curve,
                                                               max_rpm=5500,
                                                               moi=1,
                                                               damping_rate_full_throttle=0.15,
                                                               damping_rate_zero_throttle_clutch_engaged=2,
                                                               damping_rate_zero_throttle_clutch_disengaged=0.35,
                                                               use_gear_autobox=True,
                                                               gear_switch_time=0.5,
                                                               clutch_strength=5,
                                                               mass=1645,
                                                               drag_coefficient=0.3,
                                                               center_of_mass=carla.Vector3D(0.0, 0.0, 0.0),
                                                               steering_curve=[[0,1],[10, 0.5]],
                                                               wheels=wheels)
															   vehicle.apply_physics_control(physics_control)
 
# To change the variable
physics_control.torque_curve.a.x = 50
vehicle.apply_physics_control(physics_control)

Here I have a issue:
Attribute: 'list' object has no attribute 'a'

Did you have some ideia about it ?

Thank you very much again

DanielsJusts, 2019-05-14:
@DiegoOGV, you were on the right track at the beginning:

```
a = carla.Vector2D(0.0,450.0)
b = carla.Vector2D(x = 5000.0, y = 350.0)
torque_curve = [a,b]

physics_control = carla.VehiclePhysicsControl(torque_curve=torque_curve)
vehicle.apply_physics_control(physics_control)
```
These code lines will change the `torque_curve` of your vehicle to the ones you defined with variables `a` and `b`.

As for what you tried at the end - the object `physics_control` has a variable `torque_curve`, which is a list of `carla.Vector2D` objects. What you tried to do is to access attribute `a` on a list object, which doesn't exist, therefore Python threw an error `AttributeError: 'list' object has no attribute 'a'`. Either way, you cannot change the values of `torque_curve` directly as you tried in the end (for reasons that I'm not aware of, have tried this myself), but you could change all of them as you did in the beginning:
```
torque_curve = [a,b]
physics_control.torque_curve = torque_curve
```
Another option is to edit the `torque_curve`:
```
torque_curve = physics_control.torque_curve
torque_curve[0].x = 50
physics_control.torque_curve = torque_curve
```

I hope this helps!

closed 2019-05-29 by nsubiron

aakashsehgal, 2019-09-11:
I want to read position of all the 4 wheels ?

when I try `carla.WheelPhysicsControl().position`, it gives an error saying no attribute position in WheelPhysicsControl(). However this is an attribute.

aadityasp1, 2019-10-23:
hi @aakashsehgal , were you able to read positions of all 4 wheels? I'm facing the same issue.

aakashsehgal, 2019-10-23:
Hi @aadityasp1 Yes this works. You can try the following :

```
physics = ego.get_physics_control()
wheels = physics.wheels
wheel_FL = wheels[0]
wheel_FR = wheels[1]
wheels_RL = wheels[2]
wheels_RR = wheels[3]
```

this should give you the position of the wheels as  [x,y,z] coordinate. let me know if it helps :)

aadityasp1, 2019-10-23:
It works @aakashsehgal ! Thanks.  Also, do you know how to get the individual speeds of each wheel?

aadityasp1, 2019-10-24:
Well, the above formula works if the vehicle is moving on a straight road. How to deal with the turns? Because the wheel speeds are going to vary only at turns or on humps. And these are the places I want to collect the data.

XinnWang, 2021-09-15:
@aadityasp1 Have you got the individual speeds of each wheel ? i am facing the save problem now.

Zeleni9, 2021-09-20:
I found problem, the wheel's position is in the centimeters, so just convert in meters since the vehicle position is in the meters.

```python
wheel_location = wheel_location / 100  # Wheel position is in cm so need to be converted to meters
```