carla-simulator/carla #1444: Ray cluster, but does not connect to worker nodes on AWS.

Deepak3994, 2019-03-26:
I am using Ray 0.6.4 from github and python 3.5. I tried to setup ray cluster on aws using the example_full.yaml. In example_full.yaml i set "min_workers: 2" and "max_workers: 6". After the successfull launch of the head node, i did ssh to head node and tried to execute the train_a3c.py script. I got the below error, that the cluster has only 4 cpu and 1 gpu. Where as, inital "min_worker" i set for 2, so each instance has 4cpu and 1gpu, so all together i should have 12cpu and 3gpu including the head node.

I have attached a screenshot of the error below,

[image]

Deepak3994, 2019-03-28:
Solved..

closed 2019-03-28 by Deepak3994

buedaswag, 2019-05-28:
how did you solve this? im having the same problem on a kubernetes cluster on GCP.
ran ray start --redis-address 10.16.2.2:6379 on each worker but the workers are not doing anything, just the head node