carla-simulator/carla #1446: Adding new vehicle to CARLA

pparmesh, 2019-03-26:
I bought a truck from the unreal engine marketplace (https://www.unrealengine.com/marketplace/en-US/slug/truck-trailer-attach-detach-system) and now I am trying to import it to CARLA. 
1. I opened the unreal project that had the truck in it and did 'migrate assets' to the entire truck folder and I migrated it to /carla/Unreal/CarlaUE4/Content/Carla/Static/Vehicles/4Wheeled
2. I tried following "How to add assets" (https://carla.readthedocs.io/en/latest/how_to_add_assets/) and first exported the truck skeletal mesh as fbx and then imported it. However I get errors and I am not able to see the truck when I go to tune its physics. 
The asset that I bought from the marketplace already has an animation blueprint, a skeletal mesh, a physics asset etc so I'm not sure if I need to follow all the steps given in "How to add assets" .
Can someone kindly explain the procedure as I am not able to understand the documentation?
Thank you so much! 
To summarize: I have a truck vehicle that I bought from Unreal and I want to import it to Carla. Thanks!

closed 2019-06-01 by stale[bot]

journeyman9, 2019-06-03:
Have you made any progress on this? I had success with importing a skeletal mesh as .fbx, but it was for another car. I have a feeling the extra two wheels on the tractor will negatively affect the BaseVehiclePawn class for the blueprint. Perhaps the skeleton will need to be modified so the additional two wheels are grouped up with the VehicleBase and not treated as wheels.

Regarding a trailer, I'm thinking the best way to approach it may be to create it as another car with the BaseVehiclePawn for the trailer. I'm not a UE4 expert, but after playing with Carla, this is what I'd think to do.

journeyman9, 2019-06-03:
I wish it was as simple as plug and play from the Unreal Marketplace, but I think it's not easily compatible with Carla. Carla uses the BaseVehiclePawn class whose parent class is CarlaWheeledVehicle. The CarlaWheeledVehicle is using PhysXVehicleManager according to 

https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Vehicle/CarlaWheeledVehicle.cpp

The truck and trailer blueprint from the marketplace directly uses PhysXVehicles as the parent class for the tractor blueprint. I think it is possible to get things to link correctly and I am no UE4 guru, but I'm not sure if it is as simple as migrating a pre-made asset in order to get the Carla API to work with it. Can someone with more expertise answer this? I am also interested in implementing this tractor-trailer.

journeyman9, 2019-06-03:
From CarlaWheeledVehicle.cpp
```
// Wheels Setup
  const int PhysicsWheelsNum = PhysicsControl.Wheels.Num();
  if (PhysicsWheelsNum != 4)
  {
    UE_LOG(LogCarla, Error, TEXT("Number of WheelPhysicsControl is not 4."));
    return;
}
```
So it throws an error if the number of wheels does not equal 4, but there has to be an exception because motorcycles can also be used. It may take some modification of the underlying Carla .cpp files

RonGros, 2020-03-04:
I see that this case is closed. is it because there was nothing done on it or were you able to import the truck? I am actually looking into doing the exact same thing

onebitme, 2020-04-20:
@RonGros @journeyman9 dealing with a similar problem. Any guides or walkthroughs will be appreciated.

CharlesXu1124, 2022-03-21:
Why is this issue closed? I don't mind making some changes to the source file

frankeng, 2022-05-09:
Hi all,

I also want to use a truck semitrailer in to Carla and  I managed to get one  into Carla (car with a simple trailer). And I can drive around with it.

see the video https://youtu.be/nr4RuLNotFo

What I did was that I added the trailer also as a 4 wheeled car. 

[image]

And  added a car with a hitch to couple the trailer to.

[image]

With the blueprint shown in the picture I connected the car and the trailer

I put the trailer into to map  (town10) in the Unreal editor. Then I Pressed the play button in unreal and started the py-game window with the comment: "python manual_control.py --filter scar".

Where  "scar" is the simple car with the hitch. Then I drive the scar with the hitch to the trailer connection point and it will connect to the trailer (because it enters the collision box).

If it is connected I can drive around with the trailer.

I now have 2 problems: 

When I stopped the vehicle combination, sometimes it gets stuck. I then cannot move it any more or it is moving very slow. The work around is to try to drive it backwards ( it then hardly moves) and the I can drive forwards again. Does anyone know what the problem could be?

Sometimes the the combination is  launched into the air, probably I did not set the teleport correctly?

frankeng, 2022-05-10:
I solved the issue with the stuck trailer. It is getting into sleep mode, and you have to awake it:
[image]

teleport on, will solve the launching in the air problem