carla-simulator/carla #1449: Question: generating a data set of RGB + semantic_segmentation camera videos

advance512, 2019-03-26:
Hi there,

We are interested in creating a data set of videos,  being a set of pairs of corresponding videos: RGB and semantic_segmentation videos.

1. Does any open data set of CARLA videos of this kind exist?
2. If not, is it possible to generate video output to two different cameras?
3. Are there any "recordings" of CARLA scenarios and play-drives? That means, no manual or automatic controls - just replaying of drives done before by various AI and human drivers?

Thanks for your help! :)

fangweihao123, 2019-05-04:
Hi advance512,
    I haven't found any open data set of CARLA videos yet.But i write a python script for Carla0.9.3 which can capture the image from RGB_camera and transform it to video through module scikit-video,the code is in the url  https://github.com/fangweihao123/Carla_video_capture . However , i try to maintain as much   frames as possible,so the performance of my code may not be very good.It takes about 50min to get 2 min Carla video,so I will try to improve it.
    Hope my reply can be helpful for you :)

advance512, 2019-05-06:
@fangweihao123 this is awesome, thank you!

closed 2019-07-12 by stale[bot]