carla-simulator/carla #1453: Reproducability when using Synchronous mode in Carla 0.9.4

ykaganov, 2019-03-27:
I am running a python script using the Carla 0.9.4 client in synchronous mode simulating a single car with no pedestrians. I am getting * different*  results each run even though i set the VehicleRandomSeed  and this is because the traffic lights are *not * in the same state every run. Is there a way to change the state of all traffic lights so they are the same at the first world.tick() command every run? or are the traffic lights only running asynchronsouly ?

nsubiron, 2019-03-27:
The random seeds are no longer used in newer versions. Your client would need to set the state of each traffic light at the beginning.

ykaganov, 2019-03-27:
First, thank you for the prompted answer!!. Does that apply to vehicles and pedestrians as well? I was under the impression that the random seeds did work for my vehicle....

nsubiron, 2019-03-27:
Since now you can control (almost) everything from the client-side, there is (almost) no need for random numbers. The last piece of code that still uses random numbers is the autopilot code at intersections, if a vehicle is in autopilot mode it turns at intersections at random, and unfortunately there is no way now to specify that random seed. This haven't been fixed because our idea is to move the autopilot to Python too.

ykaganov, 2019-03-27:
That makes sense. Thanks !

ykaganov, 2019-03-27:
I just read about the exciting new feature of the logging and playback system !! Will the playback system record the state of traffic lights ? Will do it for pedestrian actions?

ykaganov, 2019-03-27:
and for vehicle controls /decisions made by the autopilot?

nsubiron, 2019-03-27:
I haven't had the chance to play with the playback system yet :sweat_smile: but it should record everything in the scene so when you replay you get the same result.

ykaganov, 2019-03-27:
So cool! I got to check it out,  Thanks!!

ykaganov, 2019-04-17:
In case anyone finds it useful, I managed to record the simulation and reproduce it exactly using the playback system described here:[ ](http://carla.org/2019/03/01/release-0.9.4/). For replay, switching the server to synchronous mode and then using client.replay_file seems to do the work for me.

closed 2019-04-17 by ykaganov

havefun28, 2019-07-27:
Hi @ykaganov ,
 Thanks for your post! May I ask how you switch to synchronous mode to avoid frame drop?
I am using 0.9.5 and add
```
settings = self.world.get_settings()
settings.synchronous_mode = True
```
in `manual_control.py` class `World`, and also tried `self.client.apply_batch_sync(batch)`. However, it seems that the control and measurement still differ by a few frames.

antoniohernandezm, 2021-03-09:
Hi @RuihanGao 

Could You solve the problem with sync mode in manual control?  I am triyng to perform a record using manual_control.py and sync mode , in order not to lose any image. I set sync mode adding: 

settings = self.world.get_settings()
settings.synchronous_mode = True

to class World, but when I launch the code, the simulation freezes and I have to force the program to stop.
I have checked that I receive the tick in the line world.tick(clock), I don't know why the simulation is frozen. The program works because pressing esc ends the execution of the code.

himanshuak, 2022-08-02:
> In case anyone finds it useful, I managed to record the simulation and reproduce it exactly using the playback system described here: http://carla.org/2019/03/01/release-0.9.4/. Just to make it clear, my goal was to rerun the exact same simulation but record data with a different Lidar camera. I wanted to use synchronous mode because otherwise I would have frame drops. Switching the server to synchronous mode and then using client.replay_file seems to do the work for me.

I am in a similar situation as well. I am trying to get lidar data with different lidar configurations(channels, number of lidars, position of lidar, etc) for the same scene. It would be helpful if you could post some example snippet on how to change sensor setting while replaying the simulation. Thanks in advance