carla-simulator/carla #1457: turning on and off street lights

kbuggenhout, 2019-03-27:
question : 
 I seen in the streetlight blueprints there is a spotlight defined, but when I run with dynamic weather, time elapses and some sort of night time ocurs, I would have expected street lights to come on at a certain point. ( or is this a roadmap feature) 

for some of the test runs this can influence AI decisions if based on visual rgb camera, I havent found provisioning lights in the car models, and am adding that at the moment to base vehicle, which also should turn on at night time and in tunnels ( easy to trigger) but I havent found a way to do this from dynamic_weather.py or any reference in the carla API, guess I need to dig into EU API to filter on the spotlights with streetlights as parent and turn thes visible/invisible when needed. brake lights would be just cosmetic as we use other methods to devine the action of the car in front.

I am also trying to build carla with UE 4.22 to enable real time raytracing to get closer to reality image quality. running on a workstation with a Quadro RTX 6000 quite performant.

thnx

nsubiron, 2019-04-15:
Hi @kbuggenhout,

Sorry for the late reply. Yes, we plan to add night mode in the near-ish future (a few months from now, it's not so trivial to create a good-looking night mode keeping an acceptable performance). For this we would also need to add lights to the vehicles of course. For now I'm afraid you'll need to switch lights on from the UE4 editor and create a new Carla build.

closed 2019-06-21 by stale[bot]

PatrickFSLin, 2023-09-14:
> Hi @kbuggenhout,
>
> Sorry for the late reply. Yes, we plan to add night mode in the near-ish future (a few months from now, it's not so trivial to create a good-looking night mode keeping an [...]

Hi @nsubiron 
Is there night mode supported in recent Carla versions (0.9.12, 0.9.13, 0.9.14)?
I have done some tests on two PCs, one is for Carla server and the other one is for Carla client. Both are Carla 0.9.14 running on RTX 3080
And I noticed that rendering performance of Carla server became poor if it switch sunny(>40 FPS) to night (~20 FPS).
If the feature is not supported, do you have solution for this condition?
Thanks