carla-simulator/carla #1459: destroy actor fail

zlw21gxy, 2019-03-28:
I am using carla0.9.4 with ubuntu16.04 but after running when I want to destroy actors weird things happened, if I check whether that actor alive or not it shows me actor already dead, but when python program finish running, LIFETIME warning will pop up and tell me actors still alive
[image]
[image]
[image]
[image]

nsubiron, 2019-03-29:
Hi @zlw21gxy,

I see the confusion, "LIFETIME" messages are a debug check on the actor handle, while the message _"attempting to destroy an actor that is already dead"_ refers to an actual actor in the simulation. They're not talking about the same object, you may have destroyed the vehicle in the simulator, but in Python you still have its handle object. (I think lifetime debug checks were removed in the last build).

If the LIFETIME messages only happen when you kill your script it's fine, but if they happen when exiting successfully, it may be showing that you have a memory leak. In that case, probably you are registering a callback (on "listen" or "on_world_tick") that has a reference to a Carla object (actor, world, or client). This creates a circular reference between Python objects so they're never destructed. What we do to avoid that is using weak references, for instance
```py
import weakref

class MySensorWrap(object):
    def __init__(self):
        self.sensor = world.spawn_actor(...)
        weak_self = weakref.ref(self)
        self.sensor.listen(lambda data: MySensorWrap._on_listen(weak_self, data))

    @staticmethod
    def _on_listen(weak_self, data):
        self = weak_self()
        if not self:
            return
        self.parse_data(data)
```

closed 2019-06-04 by stale[bot]

Yumin-Sun-00, 2020-03-31:
I got the same error when running the `examples/tutorial.py`

```
created vehicle.chevrolet.impala
created sensor.camera.depth
moved vehicle to Location(x=267.309509, y=-5.083644, z=-0.048065)
created vehicle.kawasaki.ninja
created vehicle.gazelle.omafiets
created vehicle.bh.crossbike
created vehicle.volkswagen.t2
created vehicle.gazelle.omafiets
created vehicle.gazelle.omafiets
created vehicle.tesla.cybertruck
created vehicle.harley-davidson.low_rider
created vehicle.bmw.grandtourer
created vehicle.carlamotors.carlacola
destroying actors
terminate called after throwing an instance of 'std::runtime_error'
  what():  trying to operate on a destroyed actor; an actor's function was called, but the actor is already destroyed.
Aborted (core dumped)
```

2000222, 2020-05-03:
> I got the same error when running the `examples/tutorial.py`
>
> ```
> created vehicle.chevrolet.impala
> created sensor.camera.depth
> moved vehicle to Location(x=267.309509, y=-5.083644, z=-0.048065) [...]

True, I have obtained the same error, have you found some solutions? Thanks.

YacineBenameur, 2020-07-22:
Hi,
Just add `ego_vehicle.set_autopilot(False)` before ` ego_vehicle.destroy()`
cheers

qhaas, 2020-11-06:
> True, I have obtained the same error, have you found some solutions? Thanks.

Same error in CARLA 0.9.8, but [said error is in the cleanup step](https://github.com/carla-simulator/carla/blob/0.9.8/PythonAPI/examples/tutorial.py#L122), so likely harmless.

SuperShinyEyes, 2021-09-17:
I got a same error in 0.9.11.