carla-simulator/carla #1460: throw what():  bad_weak_ptr when spawn camera

zlw21gxy, 2019-03-28:
Currently when I try to wrapper carla into gym env style(which means a lot reset), carla will throw what():  bad_weak_ptr unexpected while running, this error will block the whole process, but it's not always happened. sometime it will happened per hour sometime per day.
Does anyone face the same issue?
[image]
[image]

zdx3578, 2019-03-28:
camera = world.spawn_actor(camera_bp, relative_transform, attach_to=my_vehicle)

self.camera_rgb1 = world.spawn_actor(camera_rgb1, camera_transform, attach_to=self.vehicle)

closed 2019-06-03 by stale[bot]

parhartanvir, 2019-11-06:
Facing the same issue, when trying to run the ros-bridge, openai-gym style.