carla-simulator/carla #1461: Spawn Pedestrians on sidewalk and make them walk

stamatic, 2019-03-28:
I'm currently using Carla 0.9.4 (pre-compiled) on Ubuntu 18.04.2. I have been trying  to get the pedestrians to be spawned on the sidewalk and to get them to move. Both of these tasks failed. I opened the maps of Town01 and Town02 in the UnReal editor and saw that in these two scenarios there are so called _WalkerStartSpawnPoints_ . Is there a way to access these through the PythonAPI in a similar way to how one can retrieve the spawn points on the road? Looking through some older posts, I saw that in the past it was possible to set the number of pedestrians through CarlaSettings.ini, I know that this is now deprecated. So I tried to set the number of pedestrians as command line argument. When the simulations starts I can see that the parameter was read correctly, but no walkers are spawned in this way. Is this the intended behavior?

In [https://github.com/carla-simulator/carla/issues/1229](https://github.com/carla-simulator/carla/issues/1229) it was asked how to control the spawned walkers. One of the users suggested to simply set some commands, this does not really serve the purpose of the simulation. While reading the original Carla paper I came across the following paragraph: 

> Pedestrians navigate the streets according to a town-specific navigation map, which conveys location-based cost. This cost is designed to encourage pedestrians to walk along sidewalks and marked road crossings, but allows them to cross roads at any point. Pedestrians wander around town in accordance with this map, avoiding each other and trying to avoid vehicles.

How can one enforce this behavior? In [https://github.com/carla-simulator/carla/issues/258](https://github.com/carla-simulator/carla/issues/258) the navigation mesh is mentioned, however I have not been able to see figure out how to use it.

mallela, 2019-03-29:
`
        bp = random.choice(blueprint_library.filter('walker'))
        transform = world.get_map().get_spawn_points()[0]
        pedestrain = world.spawn_actor(bp, transform)

        control = carla.WalkerControl()

        while True:
            control = carla.WalkerControl()
            control.speed = 0.9
            control.direction.y = 1
            control.direction.x = 0
            control.direction.z = 0
            pedestrain.apply_control(control)
            time.sleep(1)

            # boolean operation 1 for jump and 0 for not to jump
            control.jump = 0
            pedestrain.apply_control(control)
            time.sleep(1)

            pedestrain.apply_control(control)
            time.sleep(1)

            i += 1
`

Hope that helps!

stamatic, 2019-04-01:
Hi @mallela, 

Thanks for getting back to me. Unfortunately it doesn't help. These are the things I already knew how to do. I wanted to avoid usign `world.get_map().get_spawn_points()` as it returns the spawn points on the road, which are not were I want to place my pedestrians. I ended up manually defining the spawn points by reading them from the map through the Unreal Editor. 
Also when it comes to controlling the pedestrians, as I said in my original question, I'm aware of how to enforce simple motion commands. But was trying to find out more with regards to the more automated motion described within the paper as well as in some other replies here on GitHub. Nevertheless, thanks for your answer.

jagdishbhanushali, 2019-04-02:
hi @stamatic ,
Please check https://github.com/carla-simulator/carla/issues/1229#issuecomment-478790154 and https://github.com/carla-simulator/carla/issues/1229#issuecomment-470678046

i14d4, 2019-04-24:
Hey guys,
I'm facing exactly the same issues as @stamatic. 
The workaround to spawn pedestrians on the sidewalk as described by @jagdishbhanushali  solves the problem only partially since the issue with no automated "random" pedestrian movements remains.
Is there any solution or approach how to let the pedestrians walk automated or when it will be included (again) ? 
Thanks in advance!

ruijieren98, 2019-06-13:
Hi @cstamatiadis, have you figured out how to let pedestrians walk automatically around the town? According to other comments for the former version of Carla, pedestrians can walk from set point A to B automatically using algorithm provided by Unreal Engine. However, for Carla 0.9.5, I haven't find APIs to let pedestrians walk from point A to B, I use Actor.set_destination() to try to do this but the pedestrian just teleports from spawn point to his destination. Any suggestion? Thanks.

cstamatiadis, 2019-06-13:
Hi @ruijieren98 , 
unfortunately I didn't manage to solve this problem. For my project it wasn't too important to get the pedestrian to actually walk, so I simply spawned them and kept them static.

closed 2019-08-19 by stale[bot]