carla-simulator/carla #1465: Cannot find the recording file manual_recording.rec

zhang-xinxin, 2019-03-30:
I want to test the logging and playback feature on Carla 0.9.4 pre-compiled version. I run the manual_control.py, start/stop the recording using CTRL+R, and then playback with CTRL+P. The feature works well. 

But I couldn't find the "manual_recording.rec" file in the carla root directory.  Could you tell me where was it saved? I'd like to analyze its format and create a customized .rec file to playback.  Thanks!

zhang-xinxin, 2019-04-01:
I finally found the file in /CARLA_ROOT/CarlaUE4/Saved/ directory. It's a binary file. How could I parse the file format and edit it?

bernatx, 2019-04-15:
Hi zhang-xinxin,

We have written a description of the binary file format (.LOG) used for the recorder system, so you can understand how to parse the file if required. 

It will be ready probably for the next release (0.9.6).

What is not clear is why do need to edit the file, because it may be corrupted after that.

Regards

closed 2019-07-02 by stale[bot]

0102TH, 2019-12-05:
carla 0.9.6，I found it here.
[image]

icaropires, 2021-05-01:
Just to help anyone googling, in Linux with CARLA 0.9.11, it's under `~/.config/Epic/CarlaUE4/Saved/manual_recording.rec`.

PR #1576 (2019-07-08): Bernatx/recorder doc
#### Description

Updated documentation about the recording system.

It includes a tutorial on how to use it, and also a description of the binary file format used.

#### Where has this been tested?

  * **Platform(s):** Linux 16.04
  * **Python version(s):** 2.7 , 3.7
  * **Unreal Engine version(s):** 4.21

Fixes #1465

#### Possible Drawbacks

None

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/1576)