carla-simulator/carla #1466: ModuleNotFoundError: No module named 'carla'

INT-ZC, 2019-03-30:
OS: Windwos 10 18865
Carla: 0.9.4

Hi, 

I encountered the error `ModuleNotFoundError: No module named 'carla'` when trying to execute the command python spawn_npc.py -n 80 as suggested by the document. After a little research, I become aware that I'll need to easy_install the .egg file in PythonAPI folder. However, it has a dependencies that `Processing dependencies for carla==0.9.4` which made it fail again. What shall I do next?

INT-ZC, 2019-03-30:
I double checked, it seems that I'm using Python 3.6 as CNTK does not support 3.7 yet. However, carla does not support 3.6?

slowergmajor, 2019-03-30:
You need to import the carla, try to make sure the script include the 
````
import glob
import os
import sys
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
import carla
````
part. and make sure the `'../carla/dist/carla-*%d.%d-%s.egg'` path right .
hope this will help you.

srikanth-narayanan, 2019-05-03:
In Ubuntu Bionic (18.04) and python 3.6 os.name reports as 'posix'
can we add posix as additional condition to be compatible with the latest version ?

remitoudic, 2019-06-11:
Hey Guys 
I add this comment because I spent too much  time  on this issue .I try to change the sys.path.append and experiment a couple of other things...
At the this end , I realize  that there are  2 eggs ( zip files in carla/dist )  one for python 2.7  and one for python 3.5  and therefore you  have to run the python client script with one of these distributions.  Then all is fine .... :) . 
I am pretty sure it is somewhere written in the doc ....

srikanth-narayanan, 2019-06-11:
@remitoudic - Thanks for the answer. I realized the same. I unzipped the distribution (based on my python version) I needed and placed in the my path.

closed 2019-08-17 by stale[bot]

autonomobil, 2019-08-19:
I had a similar issue in a conda enviroment. This was my solution:
 
- locate .egg-files in `Carla/PythonAPI/carla/dist/`
- change/install python version in conda enviroment which fits closest to an .egg-file (for me carla-0.9.6-py3.5-linux-x86_64.egg -> 3.5)
- delete all other .egg-files (just for tidiness)
- extract your .egg-file, e.g. carla-0.9.6-py3.5-linux-x86_64
- create `setup.py` inside the folder (carla-0.9.6-py3.5-linux-x86_64) with the following content:

```
from distutils.core import setup
setup(name='carla',
      version='0.9.6', #doesn't matter I guess
      py_modules=['carla'],
      )
```
- then you can install the carla-python-package via pip: 
`pip install -e ~/Carla/PythonAPI/carla/dist/carla-0.9.6-py3.5-linux-x86_64`

That solved it for me. I don't even need the egg-path-appending-stuff, which is happening at the beginning of every python-script, just `import carla.`

RoozbehBandpey, 2020-03-13:
You either need to write your script dirercty where you extracted carla. In that case the part ```sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg'...``` will detect carla path in your environment variables.

Or change the path in ```sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg'...``` to directrly point to where you extracted carla.

A cleaner way would be to copy ```'../carla/dist/carla-*%d.%d-%s.egg'``` in your python site pachkages, then you can import it normally anywhere you are developing your script.

anurag-176, 2020-09-12:
Did you guys check out sentdex's video?
https://www.youtube.com/watch?v=J1F32aVSYaU&list=PLQVvvaa0QuDeI12McNQdnTlWz9XlCa0uo

He uses:
`py -3.7 python spawn_npc.py -n 80`

tetsu-kikuchi, 2022-11-20:
Information as of 20 Nov. 2022.

Related documents are [this one](https://carla.readthedocs.io/en/0.9.13/build_faq/#importerror-no-module-named-carla) and [this one](https://carla.readthedocs.io/en/0.9.13/start_quickstart/#install-client-library).

In my case, carla installation worked in the following steps.  I used Ubuntu 20.04, and Carla v0.9.13. 
- Prepare Python version 3.7.xx   (I used 3.7.10)
- Upgrade pip (I used pip 22.3.1)
- $ cd PythonAPI/carla/dist/
- $ pip install carla-0.9.13-cp37-cp37m-manylinux_2_27_x86_64.whl

ishaan95, 2023-02-18:
Update as of Feb 18th 2023. 

You can directly execute `pip install carla==<the version number>` to get the python module released on [the carla pypi](https://pypi.org/project/carla/#description).