carla-simulator/carla #1468: how to fix OpenGL 4.3 not supported by driver

J-zin, 2019-03-30:
i got this problem, anybody can help me, pls~~
my gpu informance as follow:
Sat Mar 30 17:22:08 2019
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 384.81                 Driver Version: 384.81                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla K40c          Off  | 00000000:02:00.0 Off |                    0 |
| 23%   34C    P8    20W / 235W |     17MiB / 11439MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Tesla K40c          Off  | 00000000:03:00.0 Off |                    0 |
| 23%   35C    P8    22W / 235W |     10MiB / 11439MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1089      G   /usr/lib/xorg/Xorg                             6MiB |
|    0     13327      G   /usr/lib/xorg/Xorg                             9MiB |
|    1     13327      G   /usr/lib/xorg/Xorg                             9MiB |
+-----------------------------------------------------------------------------+
should i update nvidia driver? 

[2019.03.30-09.20.51:397][  0]LogInit: Error: Linux_PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
[2019.03.30-09.20.51:397][  0]LogRHI: Error: OpenGL 4.3 not supported by driver
[2019.03.30-09.20.51:398][  0]LogInit: Error: Linux_PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'

Johnnyswong, 2019-04-02:
I think reinstall the Nvidia driver will help. I have met this problem before, and I reinstall the nvidia driver, it fix the problem. My nvidia driver version is 36x.

J-zin, 2019-04-02:
@Johnnyswong 
would u connect the server without display?  i use Bitvise SHH client. and my nvidia version is  384.81. would i update it to 39x version???

Johnnyswong, 2019-04-02:
If you use the Carla container, you need to update your nvidia driver to the version higherthan 390.------------------ Original ------------------From: J-zin  <notifications@github.com>Date: Tue,Apr 2,2019 9:04 PMTo: carla-simulator/carla <carla@noreply.github.com>Cc: JunningHuang <jn8227805@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [carla-simulator/carla] how to fix OpenGL 4.3 not supported bydriver (#1468)@Johnnyswong
would u connect the server without display?  i use Bitvise SHH client. and my nvidia version is  384.81. would i update it to 39x version???

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/carla-simulator/carla/issues/1468#issuecomment-478985749",
"url": "https://github.com/carla-simulator/carla/issues/1468#issuecomment-478985749",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

J-zin, 2019-04-02:
@Johnnyswong i have been updating nvidia driver to 390.116
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.116                Driver Version: 390.116                   |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  Tesla K40c          Off  | 00000000:02:00.0 Off |                    0 |
| 24%   48C    P0   102W / 235W |   1020MiB / 11441MiB |     52%      Default |
+-------------------------------+----------------------+----------------------+
|   1  Tesla K40c          Off  | 00000000:03:00.0 Off |                    0 |
| 23%   41C    P8    23W / 235W |     10MiB / 11441MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      2448      G   /usr/lib/xorg/Xorg                            24MiB |
|    0      2474    C+G   ./CarlaUE4/Binaries/Linux/CarlaUE4           992MiB |
|    1      2448      G   /usr/lib/xorg/Xorg                             8MiB |
+-----------------------------------------------------------------------------+

but..... it still does not run.. log as follow:
[2019.04.02-14.52.07:682][  0]LogCarla:   * 14 - SoftRainSunset
[2019.04.02-14.52.07:682][  0]LogCarla: [CARLA/QualitySettings]
[2019.04.02-14.52.07:682][  0]LogCarla: Quality Settings = Epic
[2019.04.02-14.52.07:682][  0]LogCarla: [CARLA/Sensor]
[2019.04.02-14.52.07:682][  0]LogCarla: Added 0 sensors.
[2019.04.02-14.52.07:682][  0]LogCarla: Semantic Segmentation = Disabled
[2019.04.02-14.52.07:682][  0]LogCarla: ================================================================================
[2019.04.02-14.52.07:703][  0]LogWorld: Bringing up level for play took: 0.350956
[2019.04.02-14.52.07:704][  0]LogCarla: Spawning player at player start 0/152
[2019.04.02-14.52.07:705][  0]LogCarla: Changing weather settings to "ClearNoon"
[2019.04.02-14.52.07:707][  0]LogCarla: Found 152 PlayerStart positions for spawning vehicles
[2019.04.02-14.52.08:079][  0]LogCarla: Spawned all 15 vehicles
[2019.04.02-14.52.08:132][  0]LogLoad: Took 3.518708 seconds to LoadMap(/Game/Maps/Town01)
[2019.04.02-14.52.08:192][  0]LogSlate: Took 0.000052 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/DroidSansMono.ufont' (77K)
[2019.04.02-14.52.08:195][  0]LogLoad: (Engine Initialization) Total time: 4.53 seconds
[2019.04.02-14.52.08:195][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.04.02-14.52.08:276][  0]LogRenderer: Reallocating scene render targets to support 1240x900 Format 10 NumSamples 1 (Frame:1).
[2019.04.02-14.52.15:279][  0]LogContentStreaming: Texture pool size now 2000 MB
[2019.04.02-14.52.16:224][  1]LogLinux: Setting swap interval to 'Immediate'

not wrong or warming, but stop in there

J-zin, 2019-04-02:
@Johnnyswong  ...... warming: 
====================
[2019.04.02-14.52.07:703][  0]LogWorld: Bringing up level for play took: 0.350956
[2019.04.02-14.52.07:704][  0]LogCarla: Spawning player at player start 0/152
[2019.04.02-14.52.07:705][  0]LogCarla: Changing weather settings to "ClearNoon"
[2019.04.02-14.52.07:707][  0]LogCarla: Found 152 PlayerStart positions for spawning vehicles
[2019.04.02-14.52.08:079][  0]LogCarla: Spawned all 15 vehicles
[2019.04.02-14.52.08:132][  0]LogLoad: Took 3.518708 seconds to LoadMap(/Game/Maps/Town01)
[2019.04.02-14.52.08:192][  0]LogSlate: Took 0.000052 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/DroidSansMono.ufont' (77K)
[2019.04.02-14.52.08:195][  0]LogLoad: (Engine Initialization) Total time: 4.53 seconds
[2019.04.02-14.52.08:195][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.04.02-14.52.08:276][  0]LogRenderer: Reallocating scene render targets to support 1240x900 Format 10 NumSamples 1 (Frame:1).
[2019.04.02-14.52.15:279][  0]LogContentStreaming: Texture pool size now 2000 MB
[2019.04.02-14.52.16:224][  1]LogLinux: Setting swap interval to 'Immediate'
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: request failed, libcurl error: 0 (No error)
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 0 (  Trying 54.85.16.28...)
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 1 (connect to 54.85.16.28 port 443 failed: Connection timed out)
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 2 (  Trying 54.242.143.72...)
[2019.04.02-14.56.15:280][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 3 (Closing connection 0)

Johnnyswong, 2019-04-02:
The log "2019.04.02-14.52.16:224][ 1]LogLinux: Setting swap interval to 'Immediate'" means that the Carla server is ready. If you want to connect to the Carla server, you can try manual_control.py.FYI------------------ Original ------------------From: J-zin  <notifications@github.com>Date: Tue,Apr 2,2019 10:58 PMTo: carla-simulator/carla <carla@noreply.github.com>Cc: JunningHuang <jn8227805@gmail.com>, Mention <mention@noreply.github.com>Subject: Re: [carla-simulator/carla] how to fix OpenGL 4.3 not supported bydriver (#1468)@Johnnyswong  ...... warming:
[2019.04.02-14.52.07:703][  0]LogWorld: Bringing up level for play took: 0.350956
[2019.04.02-14.52.07:704][  0]LogCarla: Spawning player at player start 0/152
[2019.04.02-14.52.07:705][  0]LogCarla: Changing weather settings to "ClearNoon"
[2019.04.02-14.52.07:707][  0]LogCarla: Found 152 PlayerStart positions for spawning vehicles
[2019.04.02-14.52.08:079][  0]LogCarla: Spawned all 15 vehicles
[2019.04.02-14.52.08:132][  0]LogLoad: Took 3.518708 seconds to LoadMap(/Game/Maps/Town01)
[2019.04.02-14.52.08:192][  0]LogSlate: Took 0.000052 seconds to synchronously load lazily loaded font '../../../Engine/Content/EngineFonts/Faces/DroidSansMono.ufont' (77K)
[2019.04.02-14.52.08:195][  0]LogLoad: (Engine Initialization) Total time: 4.53 seconds
[2019.04.02-14.52.08:195][  0]LogExternalProfiler: No external profilers were discovered.  External profiling features will not be available.
[2019.04.02-14.52.08:276][  0]LogRenderer: Reallocating scene render targets to support 1240x900 Format 10 NumSamples 1 (Frame:1).
[2019.04.02-14.52.15:279][  0]LogContentStreaming: Texture pool size now 2000 MB
[2019.04.02-14.52.16:224][  1]LogLinux: Setting swap interval to 'Immediate'
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: request failed, libcurl error: 0 (No error)
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 0 (  Trying 54.85.16.28...)
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 1 (connect to 54.85.16.28 port 443 failed: Connection timed out)
[2019.04.02-14.56.15:279][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 2 (  Trying 54.242.143.72...)
[2019.04.02-14.56.15:280][249]LogHttp: Warning: 0x7f3a82b33a00: libcurl info message cache 3 (Closing connection 0)

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/carla-simulator/carla/issues/1468#issuecomment-479036588",
"url": "https://github.com/carla-simulator/carla/issues/1468#issuecomment-479036588",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

J-zin, 2019-04-02:
[image]

J-zin, 2019-04-02:
@Johnnyswong 
i run python  ./manual_control.py
but  failed to connect

Johnnyswong, 2019-04-08:
Hi, sorry for the late reply.
From the left terminal, it seems that your carla server is ready. Can you make sure that the client port matches the server port?

J-zin, 2019-04-08:
@Johnnyswong 
[image]
it seems to be succeed. but it doese not open a new console with the graphics control interface？？？
i use Bitivse SHH Client to connect my server
and the referened link as follow:
https://carla.readthedocs.io/en/latest/carla_headless/

closed 2019-06-14 by stale[bot]