carla-simulator/carla #1469: Bug in Lane_id

eslambakr, 2019-03-30:
Hello,

I think there is a bug in lane_id, as I noticed that the id changes from 3 to -3 for example for certain period then becomes 3 again!, however the car don't change it's lane.

Is this a real bug or I missed some thing?

nsubiron, 2019-03-31:
Hi @eslambakr,

It's not a bug, it's how [OpenDRIVE](http://www.opendrive.org) works. A lane id is unique within its lane section, and a road section may have several lane sections.
[image]

Our waypoints are identified by 4 numbers

* **road_id:** id of the road section (as in the XODR file).
* **section_id:** id of the lane section (not standard, we added this one for extra precision).
* **lane_id:** id of the lane (as in the XODR file), negative goes forward relative to road (arbitrary) direction, positive backwards.
* **s:** distance in meters from the beginning of the road section.

But most of the time you don't need to care about these numbers unless you're directly working with the XODR file.

Johnnyswong, 2019-04-02:
Thanks a lot !!! @nsubiron  Your explanation helps a lot.

closed 2019-06-08 by stale[bot]