carla-simulator/carla #1470: Issue spawning npcs

feng3245, 2019-03-31:
I tried to run python spawn_npc.py -n 80 and got this error
Traceback (most recent call last):
  File ".\spawn_npc.py", line 23, in <module>
    import carla
  File "D:\Downloads\Downloads\CARLA_0.9.4\PythonAPI\carla-0.9.4-py3.7-win-amd64.egg\carla\__init__.py", line 7, in <module>
  File "D:\Downloads\Downloads\CARLA_0.9.4\PythonAPI\carla-0.9.4-py3.7-win-amd64.egg\carla\libcarla.py", line 7, in <module>
  File "D:\Downloads\Downloads\CARLA_0.9.4\PythonAPI\carla-0.9.4-py3.7-win-amd64.egg\carla\libcarla.py", line 6, in __bootstrap__
  File "C:\Users\Feng\Anaconda3\lib\imp.py", line 343, in load_dynamic
    return _load(spec)
ImportError: DLL load failed: The specified module could not be found.

this seem to relate to trying to load libcarla.cp37-win_amd64.pyd

TDHTTTT, 2019-04-10:
Same issue, win build: 17763.437, Carla 0.9.4 pre-compiled, python3.7. I used virtualenv to install the python3.7 packages.

ekettenburg, 2019-04-10:
YMMV and could be installation specific but here is how I fixed this issue (with precompiled CARLA 0.9.4 on windows 10 64 bit):

I used https://github.com/lucasg/Dependencies on libcarla.cp37-win_amd64.pyd to find out what was missing. 

In my case zlib.dll was missing.

I placed a zlib.dll in the path but then it failed because it was not a 64 bit version of zlib

Next I took the 64 bit version from here: https://github.com/maxirmx/Dist_zlib

And renamed the zlib1.dll from it to zlib.dll and placed it in the path (by putting it in C:/windows/system32)

Everything worked for me after that.

feng3245, 2019-04-17:
> YMMV and could be installation specific but here is how I fixed this issue (with precompiled CARLA 0.9.4 on windows 10 64 bit):
>
> I used https://github.com/lucasg/Dependencies on [...]

I used the tool after extracting zlib apparently it needs python37 as well...

feng3245, 2019-04-17:
Tried adding python37.dll and now it complains that python37.dll is not a win32 application

feng3245, 2019-04-17:
No getting ImportError: Module use of python37.dll conflicts with this version of Python.

Why lol

closed 2019-06-23 by stale[bot]