carla-simulator/carla #1471: How to get a stable time ticks when running with multiple clients?

robinalen, 2019-04-01:
hi guys,
    I wanna make a car driving schedule(like throttle 1000ms -> steering right 500ms). I used pygame.time.get_ticks to count client running time. But the UE simulating time flow will slow down with multiple clients. As a result, my pre-scheduled vehicle always turning right/left earlier or latter and against a wall. Is there a way to solve it? thanks!

closed 2019-04-01 by robinalen

sheepwxy, 2019-04-30:
Hello!
Have you realize your driving schedule? Can you show your code detail ? I was just confused about how to use pygame.time.get_ticks in Carla.

robinalen, 2019-04-30:
Hi @sheepwxy 
I used get_ticks to count time flow in client slide and calculated the offset in the modified Controller class for each vehicle’s operation. But the processing speed in Server slide will get slower when you have opened multiple clients in the same computer. 

The pre-scheduled vehicle driving is unrepeatable and inaccurate for this method, the PID control might be a better way to control vehicles with limited computing resources.

sheepwxy, 2019-04-30:
Thanks for your reply!
Actually I try to build a micro-traffic, and I want to 4-5 vehicles start at different time,which I think may related to yours problem.Do you have any advices?

robinalen, 2019-04-30:
Hi,
4-5 clients are just about meeting the slowing threshold for my computer. If you don’t want to adjust your vehicle’s timetable every time you add or reduce clients, I recommend you split the C/S to different computers. :)