carla-simulator/carla #1472: Maps created by myself cannot be connected by scripts manual_control.py

JackRen88, 2019-04-01:
Hello,carlateam.
I create a simple map in unreal engine.In the project setting,I use the setting as follows:
[image]
by pressing the play button in the unreal engine,starting the simulation environment，and running manual_control.py,I find that the unreal engine editor crash.As mentioned above,Can't  maps created by myself  be connected by scripts manual_control.py,then control the vehicle in the map?

Sakrust, 2019-04-03:
Did you rebuild Carla (including your new map) prior to your connection attempt?

XGodina, 2019-04-16:
Does This map have opendrive?

closed 2019-06-22 by stale[bot]